ompl/ompl #679: OMPL APP: python ompl_app.py Segmentation fault: 11

oncempct, 2019-09-26:
Hi, I am having problems running the OMPL App. If possible, please contact me tonight.

When I try running the program, here is the error I receive
gui [...]$ python ompl_app.py
Segmentation fault: 11

on make VERBOSE=1 I receive this error message:

In file included from /Users/[...]/Downloads/omplapp-1.4.2-Source/ompl/demos/PlannerData.cpp:45:
/opt/local/include/boost/graph/astar_search.hpp:449:13: error: non-const lvalue reference to type 'boost::dummy_property_map' cannot bind to a value of unrelated type
      'boost::astar_visitor<boost::null_visitor>'
    >::type vis = arg_pack[_visitor | default_visitor];
            ^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/[...]/Downloads/omplapp-1.4.2-Source/ompl/demos/PlannerData.cpp:173:16: note: in instantiation of function template specialization
      'boost::astar_search<boost::adjacency_list<boost::vecS, boost::vecS, boost::bidirectionalS, boost::property<vertex_type_t, ompl::base::PlannerDataVertex *,
      boost::property<boost::vertex_index_t, unsigned int, boost::no_property> >, boost::property<edge_type_t, ompl::base::PlannerDataEdge *, boost::property<boost::edge_weight_t,
      ompl::base::Cost, boost::no_property> >, boost::no_property, boost::listS>, (lambda at /Users/[...]/Downloads/omplapp-1.4.2-Source/ompl/demos/PlannerData.cpp:174:13),
      ompl::base::Cost, boost::distance_zero_t, boost::bgl_named_params<ompl::base::Cost, boost::distance_inf_t, boost::bgl_named_params<(lambda at
      /Users/[...]/Downloads/omplapp-1.4.2-Source/ompl/demos/PlannerData.cpp:177:30), boost::distance_combine_t, boost::bgl_named_params<(lambda at
      /Users/[...]/Downloads/omplapp-1.4.2-Source/ompl/demos/PlannerData.cpp:176:30), boost::distance_compare_t, boost::bgl_named_params<boost::vector_property_map<unsigned
      long, boost::typed_identity_property_map<unsigned long> >, boost::vertex_predecessor_t, boost::no_property> > > > >' requested here
        boost::astar_search(graph, start,

1 error generated.
make[2]: *** [ompl/demos/CMakeFiles/demo_PlannerData.dir/PlannerData.cpp.o] Error 1
make[1]: *** [ompl/demos/CMakeFiles/demo_PlannerData.dir/all] Error 2
make: *** [all] Error 2

I would love to keep investigating, unfortunately, I have not much time to look into this. Thank you for your time.

mamoll, 2019-09-27:
Which version of Boost? Which compiler? This fix is needed for Boost 1.71:
https://github.com/ompl/ompl/commit/962961fb86a6395d14c35f655dc439377a0cfbec
but that may be unrelated to the Python binding error. Can you attach CMakeCache.txt from your build directory?

oncempct, 2019-09-27:
Hi Mark,

Thank you for your response. I don't have a Build directory. I notice this is something missing from other threads I followed. I downloaded this version a couple of times and followed all instructions. What did I miss? Attached is the  CMakeCache.txt from the gui directory, the directory I try to run ompl_app.py

[CMakeCache.txt](https://github.com/ompl/ompl/files/3661829/CMakeCache.txt)

mamoll, 2019-09-27:
How did you build/install the code? Did you install from source, use the install script, used MacPorts/Homebrew?

oncempct, 2019-09-27:
I used MacPorts

oncempct, 2019-09-27:
Source file:
OMPL.app 1.4.2, released Jul 29, 2019 (Zip):

https://ompl.kavrakilab.org/download.html

Script:
sudo port sync \; install ompl +app

mamoll, 2019-09-27:
What happens if you run this?:
```
/opt/local/bin/ompl_app
```

oncempct, 2019-09-27:
I receive the same error message. Notice, this is the only bin directory I am able to see
Attached is a snapshot of all folders under the source file

[image]

oncempct, 2019-09-27:
Segmentation fault: 11

oncempct, 2019-09-27:
python /opt/local/bin/ompl_app
Traceback (most recent call last):
  File "/opt/local/bin/ompl_app", line 56, in <module>
    from ompl.util import OutputHandler, useOutputHandler, LogLevel, \
ModuleNotFoundError: No module named 'ompl'

mamoll, 2019-09-27:
Can you post the output of this?:
```
/opt/local/bin/python /opt/local/bin/ompl_app
/opt/local/bin/port contents ompl
```
MacPorts should have already installed a working (binary) version of OMPL.app. You shouldn't need the source code.

oncempct, 2019-09-27:
I run /opt/local/bin/ompl_app

Here is the outcome:

[image]

Thank you very much!

This is the first time I use Python Launcher. Thanks again

mamoll, 2019-09-27:
The good news is that things seem to work like they should. You can open *.cfg files from the folder /opt/local/share/ompl/resources/3D and start planning.

closed 2019-09-27 by mamoll