ompl/ompl #681: Consider pushing a new release here?

jwhendy, 2019-10-13:
Greetings,

I'm unable to build 1.4.1 on arch linux. I also tried building 1.4.2 from [OMPL source download](http://ompl.kavrakilab.org/core/download.html) and [ros-gbp](https://github.com/ros-gbp/ompl-release/releases) without success. The bitbucket repo is [cited as being deprecated](https://bitbucket.org/ompl/ompl/src/default/), but this repo doesn't have 1.4.2.

I should have suspected this, but it still took a while to diagnose: the issue was arch's recent update to boost-1.71. I can build https://github.com/ompl/ompl/commit/962961fb86a6395d14c35f655dc439377a0cfbec but not 913540035951173909c304431365d865bf49a437. We're updating packages on arch linux and having an archive/tag.tar.gz including the boost fix would be great.

mamoll, 2019-10-17:
I tagged version 1.4.2. A new release, 1.5.0, is already in the works and will probably released within a month or so.

closed 2019-10-17 by mamoll

jwhendy, 2019-10-17:
Thank you for this. This will not help us, as arch will be broken until 962961fb86a6395d14c35f655dc439377a0cfbec is included in a release without workaround patches. This will also apply to any distro on boost 1.71.

mamoll, 2019-10-17:
You will either have to wait for the 1.5.0 release or include  962961f as a patch in your packaging infrastructure for an Arch Linux release.

jwhendy, 2019-10-17:
We are already patching. Is there an overview of what it takes for a release? Can the community support in any way?