ompl/ompl #684: Boost python error during ompl build with python bindings

minmingzhao, 2019-10-22:
Below is the stdout.  I am using Boost 1.68.0 at ubuntu 16.04 with python3.5, I guess ompl is using older Boost? How to resolve this, should I downgrade the Boost, but which version? 

Thanks! 

```bash
[ 67%] Linking CXX executable ../bin/test_2dpath_simplifying
/usr/bin/cmake: /home/mzhao8/anaconda3/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
/usr/bin/cmake: /home/mzhao8/anaconda3/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
[ 67%] Built target test_2dpath_simplifying
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
[ 60%] Built target ompl
[ 60%] Built target test_planner_data
[ 60%] Built target test_2dcircles_opt_geometric
[ 61%] Built target test_state_storage
[ 62%] Built target test_2denvs_geometric
[ 63%] Building CXX object tests/CMakeFiles/py_std_function.dir/util/test_py_std_function.cpp.o
In file included from /usr/local/include/boost/python/class.hpp:23:0,
                 from /usr/local/include/boost/python.hpp:18,
                 from /home/mzhao8/CODS/Parking/PathPlan/PathSmoother/ompl/ompl-1.4.2-Source/tests/util/test_py_std_function.cpp:37:
/usr/local/include/boost/python/object/class_metadata.hpp: In function ‘void boost::python::objects::register_shared_ptr_from_python_and_casts(T*, Bases)’:
/usr/local/include/boost/python/object/class_metadata.hpp:85:91: error: wrong number of template arguments (2, should be 1)
   python::detail::force_instantiate(converter::shared_ptr_from_python<T, boost::shared_ptr>());
                                                                                           ^
In file included from /usr/local/include/boost/python/object/class_metadata.hpp:10:0,
                 from /usr/local/include/boost/python/class.hpp:23,
                 from /usr/local/include/boost/python.hpp:18,
                 from /home/mzhao8/CODS/Parking/PathPlan/PathSmoother/ompl/ompl-1.4.2-Source/tests/util/test_py_std_function.cpp:37:
/home/mzhao8/CODS/Parking/PathPlan/PathSmoother/ompl/ompl-1.4.2-Source/src/external/boost/python/converter/shared_ptr_from_python.hpp:21:8: note: provided for ‘template<class T> struct boost::python::converter::shared_ptr_from_python’
 struct shared_ptr_from_python
        ^
In file included from /usr/local/include/boost/python/class.hpp:23:0,
                 from /usr/local/include/boost/python.hpp:18,
                 from /home/mzhao8/CODS/Parking/PathPlan/PathSmoother/ompl/ompl-1.4.2-Source/tests/util/test_py_std_function.cpp:37:
/usr/local/include/boost/python/object/class_metadata.hpp:87:89: error: wrong number of template arguments (2, should be 1)
   python::detail::force_instantiate(converter::shared_ptr_from_python<T, std::shared_ptr>());
                                                                                         ^
In file included from /usr/local/include/boost/python/object/class_metadata.hpp:10:0,
                 from /usr/local/include/boost/python/class.hpp:23,
                 from /usr/local/include/boost/python.hpp:18,
                 from /home/mzhao8/CODS/Parking/PathPlan/PathSmoother/ompl/ompl-1.4.2-Source/tests/util/test_py_std_function.cpp:37:
/home/mzhao8/CODS/Parking/PathPlan/PathSmoother/ompl/ompl-1.4.2-Source/src/external/boost/python/converter/shared_ptr_from_python.hpp:21:8: note: provided for ‘template<class T> struct boost::python::converter::shared_ptr_from_python’
 struct shared_ptr_from_python
        ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-noexcept-type’
tests/CMakeFiles/py_std_function.dir/build.make:62: recipe for target 'tests/CMakeFiles/py_std_function.dir/util/test_py_std_function.cpp.o' failed
make[2]: *** [tests/CMakeFiles/py_std_function.dir/util/test_py_std_function.cpp.o] Error 1
CMakeFiles/Makefile2:870: recipe for target 'tests/CMakeFiles/py_std_function.dir/all' failed
make[1]: *** [tests/CMakeFiles/py_std_function.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

```

mamoll, 2019-10-22:
Can you try this fix:
https://github.com/ompl/ompl/commit/bf7e121471789910e18c0e50b732f37af6169ff9

minmingzhao, 2019-10-23:
Thanks, so apparently I was having a different version ompl from automatically downloaded by sh file. So now I just did git clone from github repo and get different build error now. 
Could you please help? 

```bash
[ 62%] Building CXX object tests/CMakeFiles/test_2denvs_geometric.dir/geometric/2d/2denvs.cpp.o
[ 62%] Building CXX object tests/CMakeFiles/test_2dcircles_opt_geometric.dir/geometric/2d/2dcircles_optimize.cpp.o
/usr/bin/cmake: /home/mzhao8/anaconda3/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
[ 63%] Linking CXX executable ../bin/test_state_storage
/usr/bin/cmake: /home/mzhao8/anaconda3/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
CMakeFiles/test_state_storage.dir/base/state_storage.cpp.o: In function `Load::test_method()':
state_storage.cpp:(.text+0x4017): undefined reference to `boost::test_tools::tt_detail::report_assertion(boost::test_tools::assertion_result const&, boost::unit_test::lazy_ostream const&, boost::unit_test::basic_cstring<char const>, unsigned long, boost::test_tools::tt_detail::tool_level, boost::test_tools::tt_detail::check_type, unsigned long, ...)'
state_storage.cpp:(.text+0x4200): undefined reference to `boost::test_tools::tt_detail::report_assertion(boost::test_tools::assertion_result const&, boost::unit_test::lazy_ostream const&, boost::unit_test::basic_cstring<char const>, unsigned long, boost::test_tools::tt_detail::tool_level, boost::test_tools::tt_detail::check_type, unsigned long, ...)'
CMakeFiles/test_state_storage.dir/base/state_storage.cpp.o: In function `boost::unit_test::make_test_case(boost::function<void ()> const&, boost::unit_test::basic_cstring<char const>, boost::unit_test::basic_cstring<char const>, unsigned long)':
state_storage.cpp:(.text._ZN5boost9unit_test14make_test_caseERKNS_8functionIFvvEEENS0_13basic_cstringIKcEES8_m[_ZN5boost9unit_test14make_test_caseERKNS_8functionIFvvEEENS0_13basic_cstringIKcEES8_m]+0x82): undefined reference to `boost::unit_test::test_case::test_case(boost::unit_test::basic_cstring<char const>, boost::unit_test::basic_cstring<char const>, unsigned long, boost::function<void ()> const&)'
CMakeFiles/test_state_storage.dir/base/state_storage.cpp.o: In function `_GLOBAL__sub_I_state_storage.cpp':
state_storage.cpp:(.text.startup+0x112): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)'
state_storage.cpp:(.text.startup+0x1ba): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)'
CMakeFiles/test_state_storage.dir/base/state_storage.cpp.o: In function `void boost::serialization::throw_exception<boost::archive::archive_exception>(boost::archive::archive_exception const&)':
state_storage.cpp:(.text._ZN5boost13serialization15throw_exceptionINS_7archive17archive_exceptionEEEvRKT_[_ZN5boost13serialization15throw_exceptionINS_7archive17archive_exceptionEEEvRKT_]+0x1d): undefined reference to `boost::archive::archive_exception::archive_exception(boost::archive::archive_exception const&)'
../lib/libompl.so.1.5.0: undefined reference to `boost::system::detail::generic_category_instance'
../lib/libompl.so.1.5.0: undefined reference to `boost::archive::basic_binary_iarchive<boost::archive::binary_iarchive>::load_override(boost::archive::class_name_type&)'
collect2: error: ld returned 1 exit status
tests/CMakeFiles/test_state_storage.dir/build.make:103: recipe for target 'bin/test_state_storage' failed
make[2]: *** [bin/test_state_storage] Error 1
CMakeFiles/Makefile2:548: recipe for target 'tests/CMakeFiles/test_state_storage.dir/all' failed
make[1]: *** [tests/CMakeFiles/test_state_storage.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/cmake: /home/mzhao8/anaconda3/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
[ 63%] Linking CXX executable ../bin/test_planner_data
/usr/bin/cmake: /home/mzhao8/anaconda3/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
CMakeFiles/test_planner_data.dir/base/planner_data.cpp.o: In function `bool boost::test_tools::tt_detail::check_frwd<boost::test_tools::tt_detail::equal_impl_frwd, bool, bool>(boost::test_tools::tt_detail::equal_impl_frwd, boost::unit_test::lazy_ostream const&, boost::unit_test::basic_cstring<char const>, unsigned long, boost::test_tools::tt_detail::tool_level, boost::test_tools::tt_detail::check_type, bool const&, char const*, bool const&, char const*) [clone .constprop.658]':
planner_data.cpp:(.text+0x45f): undefined reference to `boost::test_tools::tt_detail::report_assertion(boost::test_tools::assertion_result const&, boost::unit_test::lazy_ostream const&, boost::unit_test::basic_cstring<char const>, unsigned long, boost::test_tools::tt_detail::tool_level, boost::test_tools::tt_detail::check_type, unsigned long, ...)'
CMakeFiles/test_planner_data.dir/base/planner_data.cpp.o: In function `bool boost::test_tools::tt_detail::check_frwd<boost::test_tools::tt_detail::equal_impl_frwd, unsigned int, unsigned long>(boost::test_tools::tt_detail::equal_impl_frwd, boost::unit_test::lazy_ostream const&, boost::unit_test::basic_cstring<char const>, unsigned long, boost::test_tools::tt_detail::tool_level, boost::test_tools::tt_detail::check_type, unsigned int const&, [... cut ...]
../lib/libompl.so.1.5.0: undefined reference to `boost::archive::archive_exception::archive_exception(boost::archive::archive_exception const&)'
collect2: error: ld returned 1 exit status
tests/CMakeFiles/test_2denvs_geometric.dir/build.make:103: recipe for target 'bin/test_2denvs_geometric' failed
make[2]: *** [bin/test_2denvs_geometric] Error 1
CMakeFiles/Makefile2:585: recipe for target 'tests/CMakeFiles/test_2denvs_geometric.dir/all' failed
make[1]: *** [tests/CMakeFiles/test_2denvs_geometric.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

```

mamoll, 2019-10-23:
Trying to mix and packages from your OS package manager (Boost.Python) and other package managers (like conda) is not supported. You are on your own.

closed 2019-10-23 by mamoll

minmingzhao, 2019-10-24:
Figured out. Just to have an update here, in case someone have same issue like me later. 
1. Completely reinstall Boost will get rid of those error above. 
2. After that, in case you have same error of CastXML on Ubuntu 16.04 when `make update_bindings`, as of now, CastXML binary (201602) from apt-get has a bug and need to rebuild ourselves from github repo. Building CastXML is another pain, but its Superbuild version save life.

mamoll, 2019-10-24:
Thanks for letting me know. This will download a recent binary of the castxml super build:
```
wget -q -O- https://data.kitware.com/api/v1/file/5d6f1d01d35580e6dcded6dd/download | tar zxf - -C ${HOME}
export PATH=${HOME}/castxml/bin:${PATH}
```