ompl/ompl #687: Advice on OMPL settings

alaurenzi, 2020-01-08:
Hi all, and thanks for the awesome work.
In my group we are using OMPL for whole-body motion planning applied to complex humanoid robots (say ~ 40 degrees of freedom including the floating base). 

We need to enforce contact consistency, which means that contact links must have a constant position/pose. We implemented this with constrained state spaces, and our custom IK engine is used to implement the `project()` method of `ompl::base::Constraint`. The specific constrained state space that we use is the Atlas-based (the default one was unable to solve even simple problems in our experience). 

Concerning the Atlas space, we need to "anchor" it before running the random search (right?). We do it as follows:
```c++
atlas_ss->anchorChart(start);
atlas_ss->anchorChart(goal);
```
where start and goal are start and goal states (of course! :D). 

Concerning the planner, we have had good results with RRT* and RRTConnect. Please see [this small example](https://youtu.be/U8c8J-YaznI). You can recognize (i) the start state in blue, (ii) the goal state in green, and (iii) the environment in purple. Suitable state validity checkers make sure that the robot does not collide with itself and with the environment. Finally, you can see how the constrained state space successfully keeps contacts (i.e. the feet) in a fixed pose throughout the trajectory.

However, despite some sporadic success cases, most of the time the planner takes a lot of time, even minutes, to find a solution (if any). Many times, the resulting motion is "unnatural" as in [this example](https://youtu.be/hSar0xQXJ7k), with many "redundant" motion segments.

To conclude, I would like to ask for your expert opinion on these results, e.g.
 - Do you have any tips on how to speed up the planning? I am under the impression that the constraints will affect the performance a lot, maybe more than collision/self collision avoidance. Is it enough to anchor the state space to the start and goal? Can we try some strategy to improve the sampling given out constraints?
 - Would you recommend to try other planners? Which settings do you recommend to play with, given our case scenario?
 - Also, I see from the code that we are using a `UniformValidStateSampler` with NrAttempts property set to 10000. Is the state sampler something to play with as well?

Thank you in advance for you time and effort
Arturo

mamoll, 2020-01-08:
First of all, very, very cool that you got this working on such a complex system! There are several things to try. The default settings for RRT* often result in poor convergence. You can play around a bit with its parameters (turn pruning on, for instance). There are other planners that guarantee asymptotic *near*-optimality such as SPARS, SPARStwo and SST. Note that optimality is defined by default as the path length *in joint space*. The shortest path may therefore not be the most "natural" one. Defining your optimization objective and running it with RRT* might also improve the results. Planners like KPIECE1 can be very effective for high-dimensional systems (see second video on http://ompl.kavrakilab.org/gallery.html), but requires a "reasonable" low-dimensional projection. I don't think changing the NrAttempts property will help much.

@zkingston do you have any other tips?

[For future reference, the OMPL users mailing list (http://ompl.kavrakilab.org/mailingLists.html) is probably a better place to ask questions than this bug tracker on Github.]

zkingston, 2020-01-13:
Wow!! Really cool stuff.

As for planners, you might also try lowering the range parameter for RRTConnect, or try using RRTConnect with intermediate states. Make sure you are also running simplification after you get your solution, as this can dramatically improve path quality. You might also try using path hybridization to combine multiple paths into a better solution.

There are many hyperparameters to the atlas-based space that can be tuned to possible give better performance. I'd recommend first trying to tune the exploration parameter (setExploration) to something high, e.g., 0.8. This might increase the speed that the constrained space is explored and thus give you a faster time to first solution. Other parameters you can adjust are epsilon (setEpsilon), rho (setRho) and alpha (setAlpha) which all control chart parameters ([see the paper for details](http://kavrakilab.org/publications/kingston2019exploring-implicit-spaces-for-constrained.pdf)). Try increasing these to see if they give you better performance for your problem.

You can also increase the value of delta (setDelta), which controls the step size on the manifold. If your problem is sufficiently well-conditioned, you might be able to use a higher value (e.g., 0.1). Similarly, you might try increasing lambda (setLambda) to allow for more deviation from the manifold during traversal.

You might also want to take a look at how fast a typical projection takes for your constraint. If you profile the atlas-based sampler and get some numbers on how many iterations it typically takes to successfully find a solution, you can set that on the constraint (setMaxIterations) so that you don't waste too much time on probable failures. Similarly, you should set the tolerance on the constraint (setTolerance) to something reasonable for your problem, the default of 1e-4 might be too strict for your case.

If I recall correctly, the atlas-based space actually uses the AtlasBasedStateSampler. However, the sampler probably will not have that much of an effect on performance compared to changing the various parameters of the atlas-based space, as this sampler samples from the charts in the atlas.

alaurenzi, 2020-01-13:
Thanks for the fast response, we'll give a look to your proposed solutions in the next few days!

mamoll, 2020-02-02:
Closing this issue, since it's not really about a bug in OMPL. Feel free to send follow-up messages to the ompl users mailing list.

closed 2020-02-02 by mamoll