ompl/ompl #688: RRTstar return solutions that donot include the goal state

Linjackffy, 2020-01-20:
Hi,  I meet a problem when I use RRTstar planner to find a short path between start and goal state.
It seems like that RRTstar could return a solution, but sometimes the solution does not include the goal state.
The goal state I set is <200.49 -40.62 70.57 0 0 0 1>,  
and the solution's goal state is <200.282 -63.81 -41.6539 0.122714 0.216808 0.819429>
Is there a bug? Or some parameters I need to set to avoid this problem?

Here is the code I used:

```
#include <omplapp/apps/SE3RigidBodyPlanning.h>
#include <omplapp/config.h>
#include <ompl/geometric/planners/rrt/RRTstar.h>
#include <fstream>

using namespace ompl;

int main()
{
    // plan in SE3
    app::SE3RigidBodyPlanning setup;

    // load the robot and the environment
    std::string robot_fname = std::string(OMPLAPP_RESOURCE_DIR) + "/3D/cubicles_robot.dae";
    std::string env_fname = std::string(OMPLAPP_RESOURCE_DIR) + "/3D/cubicles_env.dae";
    setup.setRobotMesh(robot_fname);
    setup.setEnvironmentMesh(env_fname);

    // define start state
    base::ScopedState<base::SE3StateSpace> start(setup.getSpaceInformation());
    start->setX(-4.96);
    start->setY(-40.62);
    start->setZ(70.57);
    start->rotation().setIdentity();

    // define goal state
    base::ScopedState<base::SE3StateSpace> goal(start);
    goal->setX(200.49);
    goal->setY(-40.62);
    goal->setZ(70.57);
    goal->rotation().setIdentity();

    // set the start & goal states
    setup.setStartAndGoalStates(start, goal);

    // setting collision checking resolution to 1% of the space extent
    setup.getSpaceInformation()->setStateValidityCheckingResolution(0.01);

    setup.setPlanner(std::make_shared<ompl::geometric::RRTstar>(setup.getSpaceInformation()));
    // we call setup just so print() can show more information
    setup.setup();
    setup.print();

    // try to solve the problem
    if (setup.solve(10)) {
        // simplify & print the solution
        setup.simplifySolution();
        setup.getSolutionPath().printAsMatrix(std::cout);
        std::ofstream ofs("path.txt");
        setup.getSolutionPath().printAsMatrix(ofs);
    }

    return 0;
}
```

mamoll, 2020-02-01:
Which version of OMPL are you using? I cannot reproduce this problem with the OMPL master branch.

Linjackffy, 2020-02-03:
The version of OMPL I used is 1.4.2.
I run this example 10 times, and found that solution didnot include the goal state for 5 times.

Linjackffy, 2020-03-03:
I found out why the solution doesnot include goal state.
The solution return was approximate.

closed 2020-03-03 by Linjackffy

mamoll, 2020-03-07:
Yes, the cast to bool from PlannerStatus (the return value from solve) is a little confusing. You can change the if statement in your code to:
```
 if (setup.solve(10) == ompl::base::PlannerStatus::EXACT_SOLUTION) ...
```
to check for an exact solution.

marcoojer, 2021-07-26:
Hi, I'm using different planners with the same start and goal pose, and when I use ```if (setup.solve(10) == ompl::base::PlannerStatus::EXACT_SOLUTION)  ``` all the RRT derived optimal planners (RRTSHARP,RRTSTAR) fail. However, all the other type of planners such as rrtConnect,Kpiece,prm*, FMT* work fine for EXACT_SOLUTION. I'm even trying with a higher solving time for those which fail. And this situation feels weird I'm missing anything like a parameter to configure for these planners?? anyone has any idea about why this is happenning?