ompl/ompl #691: OMPL Gallery Missing Images/Video

eschneider1992, 2020-01-31:
Not a code bug, just having an issue with accessing the OMPL documentation when trying to check the library out. The gallery page appears to be almost blank. I can see text and headers, but no images or video. The one exception is the bottom-most post, Class Project from COMP 450 on Path Optimization, which has two visible images. Thus far I've tried it on Chrome/Firefox (linux) and Chrome/Safari (mac). Is this happening on your end as well?

https://ompl.kavrakilab.org/gallery.html

mamoll, 2020-01-31:
Thanks for reporting the issue. The problem is that we link have insecure urls (http rather https) for the videos, so browser refuse to display the content. A simple workaround for now is to visit **http**://ompl.kavrakilab.org/gallery.html instead. I will fix this soon.

mamoll, 2020-02-02:
Fixed.

closed 2020-02-02 by mamoll