ompl/ompl #692: Segmentation fault in freestate with ubuntu16.04

GYJamor, 2020-02-14:
#0  0x0000007fac68a030 in ompl::base::CompoundStateSpace::freeState(ompl::base::State*) const ()

   from /usr/local/lib/libompl.so.16

#1  0x0000007fac861744 in ompl::geometric::RRTstar::freeMemory() () from /usr/local/lib/libompl.so.16

#2  0x0000007fac8623e4 in ompl::geometric::RRTstar::~RRTstar() () from /usr/local/lib/libompl.so.16

#3  0x0000007fae6615dc in ompl::geometric::InformedRRTstar::~InformedRRTstar() ()

mamoll, 2020-02-14:
Can you provide a bit more information that would allow me to reproduce the bug? A minimal working example would be best.

GYJamor, 2020-02-15:
> minimal
**I apply ompl in a dynamic planning environment, so it will call the ~InformedRRTstar() frequency**
**----setStart and setGoal are the dynamic updateparams function**
void omplPlanner::setStart(float x, float y, float z){
  ob::ScopedState<ob::SE3StateSpace> start(space_);
  start->setXYZ(x,y,z);
  start->as<ob::SO3StateSpace::StateType>(1)->setIdentity();
  pdef_->clearStartStates();
  pdef_->addStartState(start);
}
void omplPlanner::setGoal(float x, float y, float z){
  if(prev_goal_[0] != x || prev_goal_[1] != y || prev_goal_[2] != z)
  {
    ob::ScopedState<ob::SE3StateSpace> goal(space_);
    goal->setXYZ(x,y,z);
    prev_goal_[0] = x;
    prev_goal_[1] = y;
    prev_goal_[2] = z;
    goal->as<ob::SO3StateSpace::StateType>(1)->setIdentity();
    pdef_->clearGoal();
    pdef_->setGoalState(goal);
  }
}
void omplPlanner::setMap(std::shared_ptr<octomap::OcTree> octree){
  tree_obj_ = std::shared_ptr<fcl::CollisionGeometry>(new fcl::OcTree(std::shared_ptr<const octomap::OcTree>(octree)));
}

void omplPlanner::updateParams(const Eigen::Vector3f &pose,
                const Eigen::Vector3f &goal){
  setStart(pose.x(), pose.y(), pose.z());
  setGoal(goal.x(), goal.y(), goal.z());
  updateBounds(pose.x(), pose.y(), pose.z());
}
**//here is the Constructor function**
omplPlanner::omplPlanner(){

  Quadcopter_ = std::shared_ptr<fcl::CollisionGeometry>(new fcl::Box(3, 3, 3));
  //fcl::OcTree* tree = new fcl::OcTree(std::shared_ptr<const octomap::OcTree>(new octomap::OcTree(1.5)));
  tree_obj_ = std::shared_ptr<fcl::CollisionGeometry>(new fcl::OcTree(std::shared_ptr<const octomap::OcTree>(new octomap::OcTree(1.5))));
  space_ = ob::StateSpacePtr(new ob::SE3StateSpace());
  
  ob::ScopedState<ob::SE3StateSpace> start(space_);
  // create a goal state
  ob::ScopedState<ob::SE3StateSpace> goal(space_);
  // set the bounds for the R^3 part of SE(3)
  ob::RealVectorBounds bounds(3);
  ////0,前向 1,左右 2,上下
  bounds.setLow(0,-20);
  bounds.setHigh(0,20);
  bounds.setLow(1,-20);
  bounds.setHigh(1,20);
  bounds.setLow(2,0);
  bounds.setHigh(2,20);

  space_->as<ob::SE3StateSpace>()->setBounds(bounds);

  // construct an instance of  space information from this state space
  si_ = ob::SpaceInformationPtr(new ob::SpaceInformation(space_));

  start->setXYZ(0,0,0);
  start->as<ob::SO3StateSpace::StateType>(1)->setIdentity();
  // start.random();

  goal->setXYZ(0,0,0);
  prev_goal_[0] = 0;
  prev_goal_[1] = 0;
  prev_goal_[2] = 0;
  goal->as<ob::SO3StateSpace::StateType>(1)->setIdentity();
  // goal.random();

  // set state validity checking for this space
  si_->setStateValidityChecker(std::bind(&omplPlanner::isStateValid, this, std::placeholders::_1));

  // create a problem instance
  pdef_ = ob::ProblemDefinitionPtr(new ob::ProblemDefinition(si_));

  // set the start and goal states
  pdef_->setStartAndGoalStates(start, goal);

  // set Optimizattion objective
  pdef_->setOptimizationObjective(getPathLengthObjWithCostToGo(si_));
}
void omplPlanner::updateBounds(float x, float y, float z){

  ob::RealVectorBounds bounds(3);
  ////0,前向 1,左右 2,上下
  bounds.setLow(0,x-30);
  bounds.setHigh(0,x+30);
  bounds.setLow(1,y-30);
  bounds.setHigh(1,y+30);
  bounds.setLow(2,z-3);
  bounds.setHigh(2,z+20);

  space_->as<ob::SE3StateSpace>()->setBounds(bounds);
}
**and main part of the plan function**
bool omplPlanner::plan(){
  ros::Time tmp = ros::Time::now();
    // create a planner for the defined space
  ob::PlannerPtr plan(new og::InformedRRTstar(si_));
  //ob::PlannerPtr plan(new og::RRTConnect(si_));
    // set the problem we are trying to solve for the planner
  plan->setProblemDefinition(pdef_);

    // perform setup steps for the planner
  plan->setup();
  ob::PlannerStatus solved = plan->solve(0.1);
}

zkingston, 2023-01-24:
Closing this issue due to no follow up - if you are interested in investigating this further, please post an example that reproduces the bug that can be compiled by us - more information is needed than the example you provided.

closed 2023-01-24 by zkingston