ompl/ompl #693: Cannot run "python ./ompl_app.py"

syc7446, 2020-02-20:
I reopened this issue which was made in Bitbutcket [here](https://bitbucket.org/ompl/ompl/issues/565/cannot-run-python-ompl_apppy).

To answer your question of what is the content in the `~/Software/omplapp-1.4.2/ompl/omplapp-1.4.2-Source/build/Release/lib/` directory, the followings are what I have (this is the same content you showed except the different version numbers):
```
_app.so        libompl.so.1.4.2     libompl_app_base.so.1.4.2
_base.so       libompl.so.15        libompl_app_base.so.15
_control.so    libompl.so           libompl_app_base.so
_geometric.so  libompl_app.so.1.4.2 py_std_function.so
_tools.so      libompl_app.so.15
_util.so       libompl_app.so
```

--

Hello,

After complete installation of omplapp on a fresh machine with Ubuntu 16.04 and 16 GB memory, I got the following error: `OSError: libompl_app_base.so.15: cannot open shared object file: No such file or directory`. I added `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/Software/omplapp-1.4.2/ompl/omplapp-1.4.2-Source/build/Release/lib/` in the .bashrc file and the above error message seems to be disappeared. But now I am getting “ImportError: No module named _app“. I followed your comments from https://bitbucket.org/ompl/ompl/issues/398/cannot-run-omplapp-importerror-no-module (removing `export MAKEFLAGS="-j `nproc`"` from the installation script) and ran the installation script a few times but I still cannot solve the issue. Interestingly, I was able to get omplapp running on the machine I am working on now before reinstalling the Ubuntu so I don’t get what is happening this time.. I am attaching the [pyplusplus_app.log](https://gist.github.com/syc7446/4ddd253b4a0cd253c0f3fd1dad3f099e) and [CMakeCache.txt](https://gist.github.com/syc7446/78db2970b20f851a7e9712737aa94f47) files. Could you please help me with this? Thanks!

mamoll, 2020-02-20:
What is the contents of `~/Software/omplapp-1.4.2/ompl/omplapp-1.4.2-Source/ompl/py-bindings/ompl/app/`?

syc7446, 2020-02-20:
I only have `__init__.py` and `__init__.pyc`.

mamoll, 2020-02-20:
Does it work if you copy `_app.so  libompl_app.dylib  libompl_app_base.dylib` to that directory?

syc7446, 2020-02-20:
I don't have `libompl_app.dylib libompl_app_base.dylib`. All I have is `libompl_app.so libompl_app_base.so`.

mamoll, 2020-02-20:
ok, copy these files instead. (`dylib` is the dynamic library extension on macOS)

syc7446, 2020-02-20:
Ah I see. Now omplapp works! Thanks so much!
Let me close this issue but do you have any hypothesis on why these files were not placed in that directory?

closed 2020-02-20 by syc7446

mamoll, 2020-02-20:
No, these files should have been copied there automatically as part of the build process.