ompl/ompl #699: stll alive after 75000 seconds!

sgaabdu4, 2020-03-06:
I'm struggling with this. I've literally been running the installation file for around 24 hours now and I'm completely stuck. This is what I did so far.

1) `http://ompl.kavrakilab.org/install-ompl-ubuntu.sh #downloaded the file`
2)` chmod u+x install-ompl-ubuntu.sh #made it executable`
3) `./install-ompl-ubuntu.sh --app #installed it`

I keep on getting still alive! Please help!

`Hit:1 http://ppa.launchpad.net/noobslab/macbuntu/ubuntu xenial InRelease
Hit:2 http://ppa.launchpad.net/webupd8team/atom/ubuntu xenial InRelease                      
Hit:3 http://gb.archive.ubuntu.com/ubuntu xenial InRelease                                   
Hit:4 http://security.ubuntu.com/ubuntu xenial-security InRelease                            
Hit:5 http://gb.archive.ubuntu.com/ubuntu xenial-updates InRelease
Hit:6 http://gb.archive.ubuntu.com/ubuntu xenial-backports InRelease
Hit:7 https://brave-browser-apt-release.s3.brave.com xenial InRelease
Get:8 https://repo.skype.com/deb stable InRelease [4,486 B]
Get:9 https://repo.skype.com/deb stable/main amd64 Packages [2,255 B]
Fetched 6,741 B in 0s (8,325 B/s)     
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pkg-config is already the newest version (0.29.1-0ubuntu1).
The following packages were automatically installed and are no longer required:
  gcc-4.8-base:i386 gcc-5-base:i386 libasan0:i386 libatomic1:i386 libboost-chrono1.58.0:i386
  libboost-dev:i386 libboost-serialization1.58.0:i386 libboost-system1.58.0:i386
  libboost-thread1.58.0:i386 libboost1.58-dev:i386 libc6-dev:i386 libgcc-4.8-dev:i386
  libgomp1:i386 libitm1:i386 libquadmath0:i386 libstdc++-4.8-dev:i386 libstdc++6:i386
  linux-libc-dev:i386
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  cmake-data libboost-filesystem1.58-dev libboost-program-options1.58-dev
  libboost-program-options1.58.0 libboost-serialization1.58-dev libboost-serialization1.58.0
  libboost-system1.58-dev libboost-test1.58-dev libboost-test1.58.0 libode4
Suggested packages:
  codeblocks eclipse ninja-build libeigen3-doc libmrpt-dev
The following packages will be REMOVED
  libode4:i386 libompl-dev:i386 libompl10:i386
The following NEW packages will be installed
  cmake cmake-data libboost-filesystem-dev libboost-filesystem1.58-dev
  libboost-program-options-dev libboost-program-options1.58-dev
  libboost-program-options1.58.0 libboost-serialization-dev libboost-serialization1.58-dev
  libboost-serialization1.58.0 libboost-system-dev libboost-system1.58-dev libboost-test-dev
  libboost-test1.58-dev libboost-test1.58.0 libeigen3-dev libode-dev libode4
0 to upgrade, 18 to newly install, 3 to remove and 0 not to upgrade.
Need to get 6,285 kB of archives.
After this operation, 32.5 MB of additional disk space will be used.
Get:1 http://gb.archive.ubuntu.com/ubuntu xenial-updates/main amd64 cmake-data all 3.5.1-1ubuntu3 [1,121 kB]
Get:2 http://gb.archive.ubuntu.com/ubuntu xenial-updates/main amd64 cmake amd64 3.5.1-1ubuntu3 [2,623 kB]
Get:3 http://gb.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 libboost-system1.58-dev amd64 1.58.0+dfsg-5ubuntu3.1 [8,526 B]
Get:4 http://gb.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 libboost-filesystem1.58-dev amd64 1.58.0+dfsg-5ubuntu3.1 [46.4 kB]
Get:5 http://gb.archive.ubuntu.com/ubuntu xenial/universe amd64 libboost-filesystem-dev amd64 1.58.0.1ubuntu1 [2,826 B]
Get:6 http://gb.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libboost-program-options1.58.0 amd64 1.58.0+dfsg-5ubuntu3.1 [138 kB]
Get:7 http://gb.archive.ubuntu.com/ubuntu xenial-updates/main amd64 libboost-program-options1.58-dev amd64 1.58.0+dfsg-5ubuntu3.1 [171 kB]
Get:8 http://gb.archive.ubuntu.com/ubuntu xenial/main amd64 libboost-program-options-dev amd64 1.58.0.1ubuntu1 [2,808 B]
Get:9 http://gb.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 libboost-serialization1.58.0 amd64 1.58.0+dfsg-5ubuntu3.1 [113 kB]
Get:10 http://gb.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 libboost-serialization1.58-dev amd64 1.58.0+dfsg-5ubuntu3.1 [147 kB]
Get:11 http://gb.archive.ubuntu.com/ubuntu xenial/universe amd64 libboost-serialization-dev amd64 1.58.0.1ubuntu1 [3,026 B]
Get:12 http://gb.archive.ubuntu.com/ubuntu xenial/universe amd64 libboost-system-dev amd64 1.58.0.1ubuntu1 [2,930 B]
Get:13 http://gb.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 libboost-test1.58.0 amd64 1.58.0+dfsg-5ubuntu3.1 [188 kB]
Get:14 http://gb.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 libboost-test1.58-dev amd64 1.58.0+dfsg-5ubuntu3.1 [232 kB]
Get:15 http://gb.archive.ubuntu.com/ubuntu xenial/universe amd64 libboost-test-dev amd64 1.58.0.1ubuntu1 [2,830 B]
Get:16 http://gb.archive.ubuntu.com/ubuntu xenial/universe amd64 libode4 amd64 2:0.13.1+git20150309-2 [363 kB]
Get:17 http://gb.archive.ubuntu.com/ubuntu xenial/universe amd64 libode-dev amd64 2:0.13.1+git20150309-2 [456 kB]
Get:18 http://gb.archive.ubuntu.com/ubuntu xenial/universe amd64 libeigen3-dev all 3.3~beta1-2 [663 kB]
[... 1624 lines cut ...]
still alive after 2732.21 seconds!
still alive after 2792.37 seconds!
still alive after 2852.55 seconds!
still alive after 2913.56 seconds!
still alive after 2973.89 seconds!
still alive after 3033.95 seconds!
still alive after 3094.19 seconds!
still alive after 3155.25 seconds!
still alive after 3216.87 seconds!
still alive after 3281.52 seconds!
still alive after 3341.86 seconds!
still alive after 3402.48 seconds!
still alive after 3463.1 seconds!
still alive after 3523.4 seconds!
still alive after 3583.83 seconds!
still alive after 3644.41 seconds!
still alive after 3705.56 seconds!
still alive after 3766.05 seconds!
still alive after 3828.07 seconds!
still alive after 3888.12 seconds!
still alive after 3948.58 seconds!
still alive after 4008.62 seconds!
still alive after 4068.69 seconds!
still alive after 4129.29 seconds!
`

mamoll, 2020-03-07:
Generating python bindings can take several hours. At least 6GB of RAM are recommended.

sgaabdu4, 2020-03-07:
Hi mark, my laptop has 8GB ram and right now its been ages since I installed it. After the 77000th second, it started doing this. Please let me know if there is anything problematic with this.

WARNING: ompl::base::OptimizationObjectivePtr ompl::base::operator+(ompl::base::OptimizationObjectivePtr const & a, ompl::base::OptimizationObjectivePtr const & b) [free operator]
> warning W1052: `Py++` will not expose free operator
> "ompl::base::OptimizationObjectivePtr
> ompl::base::operator+(ompl::base::OptimizationObjectivePtr const & a,
> ompl::base::OptimizationObjectivePtr const & b) [free operator]" - all
> classes, this operator works on, are excluded.

WARNING: ompl::base::OptimizationObjectivePtr ompl::base::operator*(double weight, ompl::base::OptimizationObjectivePtr const & a) [free operator]
> warning W1052: `Py++` will not expose free operator
> "ompl::base::OptimizationObjectivePtr ompl::base::operator*(double weight,
> ompl::base::OptimizationObjectivePtr const & a) [free operator]" - all
> classes, this operator works on, are excluded.

WARNING: ompl::base::OptimizationObjectivePtr ompl::base::operator*(ompl::base::OptimizationObjectivePtr const & a, double weight) [free operator]
> warning W1052: `Py++` will not expose free operator
> "ompl::base::OptimizationObjectivePtr
> ompl::base::operator*(ompl::base::OptimizationObjectivePtr const & a, double
> weight) [free operator]" - all classes, this operator works on, are excluded.

WARNING: ompl::base::PlannerTerminationCondition ompl::base::IterationTerminationCondition::operator ::ompl::base::PlannerTerminationCondition() [casting operator]
> warning W1016: `Py++` does not exports non-const casting operators with user
> defined type as return value. This could be changed in future.

WARNING: ompl::geometric::PathGeometric [class]
> warning W1031: `Py++` will generate class wrapper - user asked to expose non -
> public member function "freeMemory"

WARNING: ompl::geometric::PathGeometric [class]
> warning W1031: `Py++` will generate class wrapper - user asked to expose non -
> public member function "copyFrom"

WARNING: ompl::geometric::SST [class]
> warning W1031: `Py++` will generate class wrapper - user asked to expose non -
> public member function "freeMemory"

WARNING: ompl::geometric::SST::Motion * ompl::geometric::SST::Motion::getParent() const [member function]
> compilation error W1005: `Py++` cannot expose function that takes as
> argument/returns instance of non-public class. Generated code will not
> compile.

WARNING: ompl::geometric::SST::Motion * ompl::geometric::SST::Witness::getParent() const [member function]
> compilation error W1005: `Py++` cannot expose function that takes as
> argument/returns instance of non-public class. Generated code will not
> compile.

WARNING: void ompl::geometric::SST::Witness::linkRep(ompl::geometric::SST::Motion * lRep) [member function]
> compilation error W1005: `Py++` cannot expose function that takes as
> argument/returns instance of non-public class. Generated code will not
> compile.

WARNING: ompl::geometric::SST::Motion * ompl::geometric::SST::selectNode(ompl::geometric::SST::Motion * sample) [member function]
> compilation error W1005: `Py++` cannot expose function that takes as
> argument/returns instance of non-public class. Generated code will not
> compile.

WARNING: ompl::geometric::SST::Witness * ompl::geometric::SST::findClosestWitness(ompl::geometric::SST::Motion * node) [member function]
> compilation error W1005: `Py++` cannot expose function that takes as
> argument/returns instance of non-public class. Generated code will not
> compile.

WARNING: ompl::base::State * ompl::geometric::SST::monteCarloProp(ompl::geometric::SST::Motion * m) [member function]
> compilation error W1005: `Py++` cannot expose function that takes as
> argument/returns instance of non-public class. Generated code will not
> compile.

WARNING: double ompl::geometric::SST::distanceFunction(ompl::geometric::SST::Motion const * a, ompl::geometric::SST::Motion const * b) const [member function]
> compilation error W1005: `Py++` cannot expose function that takes as
> argument/returns instance of non-public class. Generated code will not
[... 1490 lines cut ...]
still alive after 3335.24 seconds!
still alive after 3395.47 seconds!
still alive after 3456.63 seconds!
still alive after 3516.97 seconds!
still alive after 3577.02 seconds!
still alive after 3637.35 seconds!
still alive after 3697.79 seconds!
still alive after 3758.06 seconds!
still alive after 3818.82 seconds!
still alive after 3878.97 seconds!
still alive after 3939.18 seconds!
still alive after 3999.26 seconds!
still alive after 4059.3 seconds!
still alive after 4119.4 seconds!
still alive after 4179.7 seconds!
still alive after 4239.8 seconds!
still alive after 4300.07 seconds!
still alive after 4361.67 seconds!
still alive after 4422.44 seconds!
still alive after 4483.19 seconds!
still alive after 4544.89 seconds!
still alive after 4607.04 seconds!
still alive after 4667.92 seconds!
still alive after 4728.97 seconds!
still alive after 4791.74 seconds!

sgaabdu4, 2020-03-30:
Hi Mark, It's been over a month and I'm stuck. It's not installing on my laptop for some reason.

I'll do everything I've done so far. Please help me as soon as possible

**For Boost:** sudo apt-get install libboost-all-dev
**For cmake:** sudo apt-get -y install cmake
**For Eigen:** sudo apt-get install libeigen3-dev
**For ODE:** sudo apt-get install libode-dev
**For Py++ and GCC:** sudo -H pip install -vU pygccxml pyplusplus
Downloaded the OMPL-Master
Extracted it...
cd ompl
mkdir -p build/Release
cd build/Release
cmake ../..
make -j 4 update_bindings **THIS DOESN'T WORK**
Please help me install OMPL. I'm so confused right now

yzy244001743, 2020-10-29:
Hi，I ran into the same problem ，Is the problem solved?

mamoll, 2021-04-16:
Generating Python bindings takes a long time and uses a lot of memory. It normally takes a few hours, but if you run out of memory it can take much longer.  This is not easy to fix.

closed 2021-04-16 by mamoll

rallen10, 2021-11-04:
With all due respect, it doesn't seem appropriate to close this issue. Is there no better way to create python bindings? I have 16GB RAM and I've been waiting a few hours for the install script to complete with the `--python` flag. This is almost a non-starter for using OMPL. At least it should be noted in the [install instructions](https://ompl.kavrakilab.org/installation.html) that it can take hours to complete, even with sufficient RAM.

zkingston, 2021-11-05:
I would recommend verifying that PyPy3 is installed on your machine and is found in the CMake configuration to expedite the Python binding generation process, it will dramatically improve generation time.

OMPL is a C++ library, and Python bindings are available only because the automatic generation process is relatively simple to maintain - the bindings are not a primary focus. Changes to improve the Python binding generation would be welcome, but we simply do not have the time nor the resources to directly address this.

rallen10, 2021-11-05:
@zkingston thank you for the PyPy3 recommendation. I will give it a shot

rallen10, 2021-12-01:
@zkingston, using pypy3 certainly helped a lot. I think I got my compilation time down to <1hr (maybe 20 min, wasn't really watching closely)

Also, for anyone who comes across [my comment about using ompl in a conda environment](https://github.com/ompl/ompl/issues/699#issuecomment-961418904), I found that I didn't need to modify the install script. Instead, after OMPL and the python bindings are built, you simply point your conda environment to the ompl python bindings using a `.pth` file. Example below where I use a conda `environment.yml` file to create a env titled `my_ompl_env` and then link my ompl py-bindings to the env:

```
conda env create -f environment.yml     # creates the my_ompl_env conda environment
echo "/home/ross/ompl-1.5.2/py-bindings" >> ~/miniconda3/envs/my_ompl_env/lib/python3.9/site-packages/ompl.pth    # create path file to point pyrmm conda env to ompl py bindings
```