ompl/ompl #700: Question: Collision Checking within Motion validator versus State validator?

Levi-Armstrong, 2020-03-10:
I noticed that if you do not set the state validator function OMPL prints out a warnign that collision checking is not being performed. 

In my case I am doing collision checking in the motion validator because I am performing continuous collision checking from one state to the next which inherently includes the the start and end waypoint so I do not perform collision checking within the state validator. 

It is just a warning message, but I wanted to check if there is an issue with only performing collision checking within the motion validator and not having any collision checking within the state validator?

mamoll, 2020-03-11:
Many planners (PRM, EST, SPARS, KPIECE, STRIDE, and variants) use a ValidStateSampler to sample states that are collision-free. If you don't have a state validity checker, their behavior is undefined. I suppose you could set a dummy state validity checker that returns `true` for every state. E.g., something like this:
```
setup.setStateValidityChecker([](const ob::State *) { return true; });
```
where setup is of type `ompl::geometric::simpleSetup`.

Levi-Armstrong, 2020-03-12:
@mamoll Thank you for the quick response. I have made the recommended changes so there is always a state validity checker.

closed 2020-03-12 by Levi-Armstrong