ompl/ompl #703: Unable to compile ompl 1.4.2+ds1-5 with boost 1.7.1

shirishag75, 2020-03-23:
Hi there, 

Unable to compile ompl 1.4.2+ds1-5 with boost 1.7.1 . See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=953873 

Also sharing the log which was shared which may help you figure out what went wrong. 
[ompl.log.gz](https://github.com/ompl/ompl/files/4371929/ompl.log.gz)

mamoll, 2020-03-24:
Fix is here:
https://github.com/ompl/ompl/commit/962961fb86a6395d14c35f655dc439377a0cfbec

shirishag75, 2020-03-26:
> Fix is here:
> [962961f](https://github.com/ompl/ompl/commit/962961fb86a6395d14c35f655dc439377a0cfbec)

by mistake I wrote it 1.7.1, it's actually 1.71 for which the fix is needed. Hope the fix is still valid for the version I am quoting now.

closed 2021-01-10 by mamoll