ompl/ompl #705: Installing OMPL on Ubuntu 16.04

sgaabdu4, 2020-03-30:
Hi Mark, It's been over a month and I'm stuck. It's not installing on my laptop for some reason.

I'll do everything I've done so far. Please help me as soon as possible

**For Boost:** `sudo apt-get install libboost-all-dev`
**For cmake:** `sudo apt-get -y install cmake`
**For Eigen:** `sudo apt-get install libeigen3-dev`
**For ODE:** `sudo apt-get install libode-dev`
**For Py++ and GCC:** `sudo -H pip install -vU pygccxml pyplusplus`
Downloaded the OMPL-Master
Extracted it...
```
cd ompl
mkdir -p build/Release
cd build/Release
cmake ../..
```
`make -j 4 update_bindings` **THIS DOESN'T WORK**
Please help me install OMPL app. I want to get the examples working as I want to work on the muti-vehicle routing problem using OMPL. I'm so confused right now.

mamoll, 2020-03-30:
There is no specific support for multi-vehicle routing in OMPL.app. So you may be better off just skipping the python binding generation part (i.e., `make update_bindings`) and just build the C++ library (i.e., just type `make`).

mamoll, 2020-03-30:
duplicate of #699

closed 2020-03-30 by mamoll