ompl/ompl #706: Error when installing

sgaabdu4, 2020-03-30:
This is the error I get when installing. I'm running Ubuntu 16.04. Please let me know what to do:

[  0%] Linking CXX shared library ../../../lib/libompl.so
/usr/bin/ld: /usr/local/lib/libode.a(collision_kernel.o): relocation R_X86_64_32S against `.bss' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libode.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
ompl/src/ompl/CMakeFiles/ompl.dir/build.make:4128: recipe for target 'lib/libompl.so.1.4.2' failed
make[2]: *** [lib/libompl.so.1.4.2] Error 1
CMakeFiles/Makefile2:911: recipe for target 'ompl/src/ompl/CMakeFiles/ompl.dir/all' failed
make[1]: *** [ompl/src/ompl/CMakeFiles/ompl.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

mamoll, 2021-01-10:
The fix is to recompile ODE with the `-fPIC` flag.

closed 2021-01-10 by mamoll

jih189, 2022-08-20:
how to recompile ODE with the -fPIC flag?