ompl/ompl #707: Error: Building package ompl[app]failed with: BUILD_FAILED. On windows 10 with vcpkg

ChaochaoLikeNoodles, 2020-04-13:
Hi, I have a problem when installing ompl[app] on Windows 10 with vcpkg. And I already submitted an issue ticket at vcpkg, please see the following link.
https://github.com/microsoft/vcpkg/issues/10753

Looks like it is related to the Eigen3 lib, but I already have Eigen3 installed. It automatically installed after I run the .\vcpkg.exe install ompl[app] command, it listed 100 libs need to be installed, and Eigen3 is one of them, and it shows Eigen3 finished installation. Maybe some path is wrong? Thank you very much for any help.

Log files show:
LINK : fatal error LNK1104: cannot open file 'Eigen3::Eigen.lib'
ninja: build stopped: subcommand failed.

gautz, 2020-05-05:
Same here with ``vcpkg install ompl[app]:x64-windows`` which equals ``ompl[app,core]:x64-windows``

Here is the log for the build
```
Starting package 100/100: ompl:x64-windows
Building package ompl[app,core]:x64-windows...
-- Downloading https://bitbucket.org/ompl/ompl/downloads/omplapp-1.4.2-Source.t
-- Extracting source D:/vcpkg/downloads/omplapp-1.4.2.tar.gz
-- Applying patch fix-findeigen3.patch
CMake Error at scripts/cmake/vcpkg_apply_patches.cmake:51 (message):
  Applying patch failed.  Checking patch CMakeLists.txt...

  error: while searching for:

  set_package_properties(Eigen3 PROPERTIES

      URL "http://eigen.tuxfamily.org"
      PURPOSE "A linear algebra library used throughout OMPL.")

  find_package(Eigen3 REQUIRED)

  include_directories("${EIGEN3_INCLUDE_DIR}")

  set_package_properties(MORSE PROPERTIES

  error: patch failed: CMakeLists.txt:110

  error: CMakeLists.txt: patch does not apply

Call Stack (most recent call first):
  scripts/cmake/vcpkg_extract_source_archive_ex.cmake:134 (vcpkg_apply_patches)
  ports/ompl/portfile.cmake:18 (vcpkg_extract_source_archive_ex)
  scripts/ports.cmake:90 (include)

Error: Building package ompl:x64-windows failed with: BUILD_FAILED
Please ensure you're using the latest portfiles with `.\vcpkg update`, then
submit an issue at https://github.com/Microsoft/vcpkg/issues including:
  Package: ompl:x64-windows
  Vcpkg version: 2020.02.04-nohash
```

Interestingly ``ompl:x64-windows`` which equals ``ompl[core]:x64-windows`` is building properly

mamoll, 2020-05-05:
Building the current master branch of omplapp from source using vcpkg for dependencies seems to work fine, so perhaps his has already been fixed:

https://ci.appveyor.com/project/mamoll/omplapp
https://ci.appveyor.com/project/mamoll/ompl

cramke, 2020-05-08:
I ran into the same problem just now. The console output is the same, only I used:
  Package: ompl:x86-windows
  Vcpkg version: 2020.02.04-nohash

closed 2023-01-19 by zkingston