ompl/ompl #708: `exactSolnPathTerminationCondition` does not work for most planners.

rconnorlawson, 2020-04-16:
I discovered this bug while attempting to study initial path quality and time to initial solution with `control:SST`, but it is applicable to almost all planners (see list below).

**Context** 

OMPL 1.4.2 includes a termination condition to stop solving immediately upon finding the first exact solution. The definition, from `src/ompl/base/PlannerTerminationCondition.h":

```c++
/** \brief Return a termination condition that will become true as soon
 * as the problem definition has an exact solution */
PlannerTerminationCondition exactSolnPlannerTerminationCondition(
        const ompl::base::ProblemDefinitionPtr &pdef);
```

**Observed issue** 

Control-based SST does not terminate when this condition is used, on a problem known to be solvable within 5 seconds when using a timed condition.

**Root cause** 

The implementation of this condition is straightforward, querying `pdef->hasExactSolution()`. 

However, _most planners don't register solutions with the `ProblemDefinition` while the planner terminal condition is in effect,_ meaning this condition is vacuously true for most planners.

The only way that `hasExactSolution()` can return true is by the planner calling `pdef->addSolutionPath()` with an exact solution. Most planners don't do this until the very end of `solve()`, after the termination condition is no longer in effect. This is a reasonable behavior for the planner, since publishing a sequence of improving paths during solve is likely undesired in most use cases and would be prohibitively expensive.

**Potential workarounds**

1. The user creates a custom `PlannerTerminalCondition` for each planner that queries the planner itself to determine whether a solution has been found, if possible. (An API for this is not present on the `Planner` interface.)
2. The user adds code to the planner implementation directly, recompiling OMPL from source to use it.

**Affected planners**

I compiled this list by `grep`ing in `src` for calls to `addSolutionPath`. Planners on the list only call `addSolutionPath` after the terminal condition is no longer in use (i.e. outside the main loop of the algorithm, typically toward the end of `solve`).

Geometric:
* SST
* ProjEST
* EST
* PDST
* LBTRRT
* TRRT
* LazyLBRRT
* RRT
* VFRRT
* pRRT
* RRTstar
* RRTXstatic
* BITstar
* ThunderRetrieveRepair
* LightningRetrieveRepair
* STRIDE
* FMT
* BFMT
* SPARS
* LazyPRM
* SPARStwo
* SBL
* KPIECE1

Control:
* SST
* Syclop
* LTLPlanner
* EST
* PDST
* RRT
* KPIECE1

mmurooka, 2021-04-04:
I had the same problem with RRT and RRTstar (`exactSolnPathTerminationCondition` always returns false).

jih189, 2022-11-02:
I think this bug still exists in the ompl code, but nobody fix it at all.

zkingston, 2023-01-20:
Closing this issue in favor of #874 - using `getIntermediateSolutionCallback` and having the termination condition check for this instead would be better than the final solution.

closed 2023-01-20 by zkingston