ompl/ompl #712: Potential Bug in RRTConnect (intermediate states)

sniyaz, 2020-04-28:
Hi there! I believe that there is a bug in the `growTree` method of `RRTConnect` where an invalid path is added to the tree in the event that the goal tree is being expanded and `addIntermediateStates_` is set to true.

https://github.com/ompl/ompl/blob/003fc6f29da3b44f9406193fe90905b6d3933f51/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp#L150

The current logic flips the direction of the path being extended based on the current tree. I believe that the current behavior is correct when `tgi.start` is true. However, when `tgi.start` is false, the direction of the motion is flipped: instead of going from the nearest-neighbor in the goal tree (`nmotion->state`) towards the target state (`dstate`), the configurations will be returned in the opposite direction from `dstate` towards `nmotion->state`.

I've drawn a few examples out by hand, and this seems incorrect to me. In the event that `tgi.start` is false, the first motion added to the tree by the loop on line `159` corresponds to `dstate`, which immediately has `nmotion` as its parent. This shortcuts all of the intermediate states between them, which are also added backwards by the same loop.

I think the fix here would actually be not having a special case on line 150, but instead always extending from `nmotion->state` towards the target state `dstate`, which is currently done only if  `tgi.start` is true.

Thanks for your time OMPL dev team!

scottpaulin, 2020-08-20:
Hey @sniyaz. I'm not a core dev, but I will have a stab at answering why the special case exists.

I agree that literature and textbooks generally talk about growing the goal tree toward new states. This does make the special case a little confusing:
https://github.com/ompl/ompl/blob/003fc6f29da3b44f9406193fe90905b6d3933f51/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp#L150

There is a good reason for it. OMPL supports asymmetric state spaces, where collision checking or interpolating the motion from state A to state B might return different results to collision checking or interpolating in the reverse direction from state B to state A. To ensure that returned paths are valid in asymmetric state spaces OMPL needs to perform all motion collision checking and interpolation in the same direction as the robot will follow along the path. For a startTree this means collision checking in the direction from the startTree to the new state. For the goal tree this means collision checking and interpolating from the new state toward the goalTree.

sniyaz, 2020-08-20:
Thanks for clearing that up @scottpaulin! That makes a lot more sense now. I totally agree that the asymmetric argument makes sense, and I now understand why this special case exists in the first place.

I think the thing I'm still concerned about is the logic on line 159:
https://github.com/ompl/ompl/blob/003fc6f29da3b44f9406193fe90905b6d3933f51/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp#L159

It doesn't seem like this loop accounts for the fact that the interpolation was indeed done backwards on line 150 (in the case that we're using the goal tree). Regardless of what direction the interpolation was done in, the states get added to either the start/goal tree in the same fashion, and (I *think*) that this is what creates the bug in the case of the goal tree.

To think through this a bit more explicitly, we interpolate from `dstate` -> `nmotion` when using the goal tree with intermediate states. That means that `states[1]` (on the first iteration of line 159) gets snapped to `nmotion` (on line 163). However, `states[1]` in this case is actually the state farthest on the path from `nmotion` (since we interpolate from `dstate` -> `nmotion`). The correct state to snap to `nmotion` on the first iteration of the loop would (I _think_) be the second-to-last member of `states`, not the second member.

I may be totally off the mark here, so apologies if I am, and thank you for your patience!

PR #970 (2023-01-24): Fix for RRTConnect with intermediate states
Fixes #712 - adding states should be reversed when extending goal tree.
closed 2023-01-24 by zkingston