ompl/ompl #713: How do I extract edges from PlannerData in Python

elishafer, 2020-05-01:
I'd like to extract the edges for a vertex using the PlannerData class in Python.
I tried using the following:
```
 #...
if solved:
    pd = ob.PlannerData(ss.getSpaceInformation())
    ss.getPlannerData(pd)
    pd.computeEdgeWeights()
    edge_list = []
    pd.getEdges(0, edge_list)
```
I get the following  error:
```
PlannerData.getEdges(PlannerData, int, list)
did not match C++ signature:
    getEdges(ompl::base::PlannerData {lvalue}, unsigned int v, std::map<unsigned int, ompl::base::PlannerDataEdge const*, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, ompl::base::PlannerDataEdge const*> > > {lvalue} edgeMap)
    getEdges(ompl::base::PlannerData {lvalue}, unsigned int v, std::vector<unsigned int, std::allocator<unsigned int> > {lvalue} edgeList)
```
Is there a way to do this? What am I doing wrong?

Thanks!

mamoll, 2020-05-01:
Try this:
```
if solved:
        pd = ob.PlannerData(ss.getSpaceInformation())
        ss.getPlannerData(pd)
        pd.computeEdgeWeights()
        edge_list = ou.vectorUint()
        pd.getEdges(0, edge_list)
        print([e for e in edge_list])
```

closed 2020-05-01 by mamoll

elishafer, 2020-05-03:
Thanks Mark!
I see that there are other definitions in the util as well. Is there any documentation on this other than on https://ompl.kavrakilab.org/python.html ?