ompl/ompl #717: segmentation fault in python

orenpeer12, 2020-05-17:
Hi all,
I'm working on Ubuntu 18.04, with python 3.6.

The following code crushes with exit code 139 (seg-fault?!)
The crush is in the last line when creating the setup...
any ideas??

Thanks!!

    space = ob.RealVectorStateSpace(2)
    bounds = ob.RealVectorBounds(2)
    bounds.setLow(0)
    bounds.setHigh(1)
    space.setBounds(bounds)
    cspace = oc.RealVectorControlSpace(space, 1)
    cbounds = ob.RealVectorBounds(1)
   cbounds.setLow(-1)
    cbounds.setHigh(1)
    cspace.setBounds(cbounds)
    setup = oc.SimpleSetup(cspace)

elishafer, 2020-06-03:
It seems that this issue happens when running the code in a debugger line by line (happened to me using PyCharm). Also, if adding a print statement to the code above `print(setup)` would also give the same exit code.
```
space = ob.RealVectorStateSpace(2)
bounds = ob.RealVectorBounds(2)
bounds.setLow(0)
bounds.setHigh(1)
space.setBounds(bounds)
cspace = oc.RealVectorControlSpace(space, 1)
cbounds = ob.RealVectorBounds(1)
cbounds.setLow(-1)
cbounds.setHigh(1)
cspace.setBounds(cbounds)
setup = oc.SimpleSetup(cspace)
print(setup)
```

zkingston, 2023-01-19:
The issue is with `SimpleSetup` not yet being setup. If `setup()` is called, the segfault disappears, as the default state propagator is set, which is the source of the segfault (control::SpaceInformation::canPropagateBackwards() being called in the print).

PR #975 (2023-02-07): Existence check to prevent segfault on uninitialized print in control::SpaceInformation
Fixes #717
closed 2023-02-07 by zkingston