ompl/ompl #725: Could not build OMPL with Python bindings on Raspberry Pi 4

rkoptev, 2020-06-02:
Hello!

I want to use OMPL with Python on [Raspberry Pi 4](https://www.raspberrypi.org/products/raspberry-pi-4-model-b/) mini computer. As far as I understood, the only way to do it - is to compile it manually, as prebuilt binaries (ros-melodic-ompl) come without python bindings.

So I'm trying to build OMPL for use on Raspberry Pi 4 in a Docker container, but experiencing some troubles. I'm using ros:melodic Docker image and `install-ompl-ubuntu.sh` script with `--python` option.

The issue is that when install script reaches `make update_bindings` command it exits with a following error:
`make: *** No rule to make target 'update_bindings'.  Stop.`

And this, I believe, is caused by the fact that during execution of cmake, following message was displayed to me:
```
-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., http://ompl.kavrakilab.org).
 * Python, <http://python.org>
   Used for python bindings.
```

I noticed that Python for some reason is missing. But I didn't noticed any failure on dependencies installation stage. python3-dev is present on the system:
```
root@9cfa79289051:~# apt list --installed | grep python3

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libpython3-dev/bionic-updates,now 3.6.7-1~18.04 armhf [installed,automatic]
libpython3-stdlib/bionic-updates,now 3.6.7-1~18.04 armhf [installed,automatic]
libpython3.6/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1 armhf [installed,automatic]
libpython3.6-dev/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1 armhf [installed,automatic]
libpython3.6-minimal/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1 armhf [installed,automatic]
libpython3.6-stdlib/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1 armhf [installed,automatic]
python3/bionic-updates,now 3.6.7-1~18.04 armhf [installed,automatic]
python3-asn1crypto/bionic,now 0.24.0-1 all [installed,automatic]
python3-cffi-backend/bionic,now 1.11.5-1 armhf [installed,automatic]
python3-crypto/bionic,now 2.6.1-8ubuntu2 armhf [installed,automatic]
python3-cryptography/bionic-updates,bionic-security,now 2.1.4-1ubuntu1.3 armhf [installed,automatic]
python3-dbus/bionic,now 1.2.6-1 armhf [installed,automatic]
python3-dev/bionic-updates,now 3.6.7-1~18.04 armhf [installed]
python3-distutils/bionic-updates,now 3.6.9-1~18.04 all [installed,automatic]
python3-gi/bionic-updates,now 3.26.1-2ubuntu1 armhf [installed,automatic]
python3-idna/bionic,now 2.6-1 all [installed,automatic]
python3-keyring/bionic,now 10.6.0-1 all [installed,automatic]
python3-keyrings.alt/bionic,now 3.0-1 all [installed,automatic]
python3-lib2to3/bionic-updates,now 3.6.9-1~18.04 all [installed,automatic]
python3-minimal/bionic-updates,now 3.6.7-1~18.04 armhf [installed,automatic]
python3-numpy/bionic,now 1:1.13.3-2ubuntu1 armhf [installed]
python3-pip/bionic-updates,now 9.0.1-2.3~ubuntu1.18.04.1 all [installed]
python3-pkg-resources/bionic,now 39.0.1-2 all [installed,automatic]
python3-secretstorage/bionic,now 2.3.1-2 all [installed,automatic]
python3-setuptools/bionic,now 39.0.1-2 all [installed,automatic]
python3-six/bionic,now 1.11.0-2 all [installed,automatic]
python3-wheel/bionic,now 0.30.0-0.2 all [installed,automatic]
python3-xdg/bionic,now 0.25-4ubuntu1 all [installed,automatic]
python3.6/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1 armhf [installed,automatic]
python3.6-dev/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1 armhf [installed,automatic]
python3.6-minimal/bionic-updates,bionic-security,now 3.6.9-1~18.04ubuntu1 armhf [installed,automatic]
```

Unfortunately, from this point I don't really know what to do. I tried same process on a regular amd64 laptop in the same docker image and it worked well. Could you please suggest on how to resolve rised issue?

Full output of `install-ompl-ubuntu.sh --python`:
```
[... 464 lines cut ...]

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., http://ompl.kavrakilab.org).
 * Python, <http://python.org>
   Used for python bindings.

-- The following Python modules were found:

 * pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /root/ompl-1.4.2-Source/build/Release
make: *** No rule to make target 'update_bindings'.  Stop.
```

rkoptev, 2020-06-03:
Update:

After researching on the issue I found that in CMakeModules/FindPython.cmake at line 86 (line 80 in latest master) there are predefined file path suffixes for python libraries for a few CPU architectures - x86_64 and i386:
`PATH_SUFFIXES "" "python${PYTHON_VERSION}/config" "x86_64-linux-gnu" "i386-linux-gnu"`

So I assume, that python bindings have never been built before for ARM, at least for version >1.4.2 . Please, correct me if I'm wrong. Now I'm wondering if that's at all possible to do?

Anyway, after appending this list with a proper suffix for RPi - `arm-linux-gnueabihf` the cmake issue has been resolved and it managed to detect my python libraries.

Finally, I can run `make update_bindings`, but another error occurred during generation:
```
$ make update_bindings
[  0%] Preparing C++ header file for Python binding generation for module control
[  0%] Built target control.h
[ 25%] Preparing C++ header file for Python binding generation for module util
[ 25%] Built target util.h
[ 25%] Creating C++ code for Python module util (see pyplusplus_util.log)
ERROR error occured, while parsing element with name "Field" and attrs "['name', 'access', 'context', 'offset', 'type', 'id']".
Error: 'file'.
Error: can't generate code for module util
```

So far, this is the farthest point where I could get. I will continue to debug further. Meanwhile, I will be extremely grateful for any advice and suggestions.

mamoll, 2020-06-04:
I have never built OMPL for Raspberry Pi. If you ultimate goal is to use OMPL within ROS, then you don't really need the Python bindings. Are you sure you want them?

rkoptev, 2020-06-07:
Yes, my ultimate goal is to use OMPL within ROS and some other Python code and structures that I already have. I understand that I can use C++ API, and especially on Raspberry Pi it will be more preferable due to processing power limitations. But if there is a chance to run the library in Python - I would like to do it.

chazyman, 2020-06-30:
So I also need to compile from source OMPL with python binding and ROS 1 on an Nvidia Jetson Nano  (aarch64-linux-gnu) for MusHr.io open source ROS 1 

I also am getting the exact same error
ERROR error occured, while parsing element with name "Field" and attrs "['name', 'access', 'context', 'offset', 'type', 'id']".
Error: 'file'.
Error: can't generate code for module util

I have not installed Spot from https://www.lrde.epita.fr/   ?   or is this error due to different dependency or setup?

Thanks for your help

mamoll, 2020-07-01:
This may be an issue with CastXML, the low-level tool used to create XML representations of the C++ code which are used to programmatically generate the Python bindings. You need a fairly recent version of CastXML. You can build it yourself and see if that fixes these issues. This repo makes it easier to build your own CastXML: https://github.com/CastXML/CastXMLSuperbuild

rkoptev, 2020-07-03:
@mamoll, indeed, castxml obtained from APT is old - 0.1-g927c739. But this is also the case for x86_64. To test your assumption, I built castxml 0.3.4 on Raspberry Pi natively and added it to PATH. And results are the same as before.

chazyman, 2020-07-03:
I also compile the superbuild of castxml on my jetson nano and i am still getting the same error. Lots in common between two different hardware.
should try re installing pygccxml from source as well ?

rkoptev, 2020-07-15:
@mamoll , do you think it's possible to compile bindings for ARM? If so, in what direction do you suggest I and @chazyman need to go?

chazyman, 2020-07-15:
Looks like I figured it out but i need to reproduce to make sure. I try latest source of CastXML not the Super. I think the pb is you need your system to use the Default version of LLVM/Clang loaded on your board. Nano sdk has a lot of stuff built-in so you need as i have learned to not deviate too much or it goes out wack quicly.

So in my case the version is 9 and you need to apt-get both llvm/clang main and dev packages for both ( so  4 packages)

Then build castxml and add OMPL dependencies but use New castxml and load apt-get gccxmlpy ...
Got binding to work in python 3 sesion ---> import ompl

mamoll, 2020-07-16:
Thanks a lot @chazyman! I know that generating the python bindings is painful. I'd be happy to add a Rasperry Pi / ARM tab on http://ompl.kavrakilab.org/installation.html. The source for that is available at https://github.com/ompl/ompl/blob/master/doc/markdown/installation.md. A PR with instructions along the lines of what you wrote to help people build OMPL + Python bindings on ARM would be much appreciated.

fmallaba, 2020-10-27:
Hi @chazyman , I also try to build python bindings on Jetson Nano and I try to follow your instructions. I built CastXML from source using clang-9 and got CastXML version 3.2. I installed pygccxml using `pip3 install pygccxml` command. But I still got the same error:

`#:~/ompl/ompl-1.5.0/build/Release$ cmake ../.. -DPYTHON_EXEC=/usr/bin/python3
-- The CXX compiler identification is Clang 9.0.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'ode'
--   Found ode, version 0.14
-- Checking for module 'flann'
--   No package 'flann' found
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Found PY_numpy: /usr/lib/python3/dist-packages/numpy  
-- Found PY_pyplusplus: /usr/local/lib/python3.6/dist-packages/pyplusplus  
-- Found PY_pygccxml: /home/fabpilot/.local/lib/python3.6/site-packages/pygccxml  
-- Found castxml: /home/fabpilot/ompl/CastXML/build/bin/castxml  
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.5.2
-- Found yaml-cpp: yaml-cpp  
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * Python, <https://www.python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/fabpilot/ompl/ompl-1.5.0/build/Release
#:~/ompl/ompl-1.5.0/build/Release$ make update_bindings
Scanning dependencies of target tools.h
[ 33%] Preparing C++ header file for Python binding generation for module tools
[ 33%] Built target tools.h
Scanning dependencies of target control.h
[ 33%] Preparing C++ header file for Python binding generation for module control
[ 33%] Built target control.h
Scanning dependencies of target geometric.h
[ 33%] Preparing C++ header file for Python binding generation for module geometric
[ 33%] Built target geometric.h
Scanning dependencies of target base.h
[ 33%] Preparing C++ header file for Python binding generation for module base
[ 33%] Built target base.h
Scanning dependencies of target util.h
[ 66%] Preparing C++ header file for Python binding generation for module util
[ 66%] Built target util.h
Scanning dependencies of target update_util_bindings
[ 66%] Creating C++ code for Python module util (see pyplusplus_util.log)
ERROR error occured, while parsing element with name "Field" and attrs "['id', 'name', 'type', 'context', 'access', 'offset']".
Error: 'line'.
Error: can't generate code for module util`

Perhaps you can tell me what am I doing wrong?

Diengoumar, 2021-02-18:
i have install pyplusplus pygccxml  , I have this error when i want to install ompl with this command:
./install-ompl-ubuntu.sh --python
-- The following Python modules were found:

 * pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/robot/ompl-1.5.2/build/Release
Python binding generation is very memory intensive. At least 6GB of RAM is recommended.
Proceeding with binding generation using 1 core...
make: *** No rule to make target 'update_bindings'.  Stop.

tuanhe, 2023-01-03:
Is there anyone solved the issue ? I tried may solution but all failed

tuanhe, 2023-01-03:
> Looks like I figured it out but i need to reproduce to make sure. I try latest source of CastXML not the Super. I think the pb is you need your system to use the Default version of LLVM/Clang loaded [...]

Hi, @chazyman  could you kindly share your solution steps to us , it will be highly appreciated

chazyman, 2023-01-03:
I have not worked on the project in a while and I was using Jetson nano not a Pi.

As I recall you need to add a path for llvm ---> LLVM_CONFIG

Look at following for inspiration ( ibelieve llvm 9 is what I used to match the jetson default platform

https://github.com/dusty-nv/jetson-containers/blob/be0dca3d19b30e64d129c92e93425c9ede40d65f/Dockerfile.ml#L72
<https://github.com/dusty-nv/jetson-containers/blob/be0dca3d19b30e64d129c92e93425c9ede40d65f/Dockerfile.ml#L72>

https://learninone209186366.wordpress.com/2019/07/24/how-to-install-the-librosa-library-in-jetson-nano-or-aarch64-module/

Hope this help out

C

zkingston, 2023-01-20:
See discussion in #976 #958 for reference - I believe OMPL with Python bindings builds on ARM platforms now.

closed 2023-01-20 by zkingston