ompl/ompl #728: Test failure: test_2denvs_geometric

wbthomason, 2020-06-07:
I consistently get a failure for test 14, `test_2denvs_geometric`, when running `make test` with OMPL v1.5.0. All other tests pass. Test 14 seems to run for a long time (~130 seconds on my machine), then fail without reporting additional information.

I’m on Arch Linux, with Boost v1.72.0. Happy to report additional information as needed - in particular, is there a way to enable failure cause information output for the tests?

mamoll, 2020-06-07:
This test can fail if the planners run too slow to solve some basic test problem within the test limits. It's possible your computer is a bit older, but perhaps you have compiled your code in Debug mode, which would also make the runs quite slow. The most recent build on the Travis CI build bot show this test completing in 43 seconds: https://travis-ci.org/github/ompl/ompl/jobs/694604283.

closed 2020-06-07 by mamoll

wbthomason, 2020-06-07:
Unfortunately, I'm compiling in release mode on a modern, fast computer. Is there anything I can do to get output from the test to debug the slowness?

mamoll, 2020-06-07:
You can run `./bin/test_2denvs_geometric` directly in your build directory.

wbthomason, 2020-06-08:
Thanks. It appears that PRM and PRMstar cause the failures - average time too long for 2D circles (both) and 2D map (PRMStar). The average times are around 0.28 for PRM, 0.24 for PRMstar map test, and 0.31 for PRMstar circles test.

My build matches the setup from https://github.com/ompl/ompl/blob/master/.travis.yml *except* that my versions of Eigen and Boost are more recent (v3.3.7 vs v 3.3.4 for Eigen, v1.65 vs v1.72 for Boost). However, I would not expect this to cause what appears to be a ~5-8x slowdown from expected performance (judging by the max average time values of 0.04 in the test definitions).

I will note that I get a large number of warnings related to Eigen types during build, which (judging by https://travis-ci.org/github/ompl/ompl/jobs/694604283) you do not. I haven't read through these in depth yet, but perhaps Eigen v3.3.7 *does* cause this problem?

Sorry for all the complication, and thank you for the help. Is there anything you would suggest I look into to figure out why PRM/PRMstar are so slow on my system?

I will fork OMPL and try building it in Travis with Ubuntu Eoan or Focal, both of which have Eigen v3.3.7, to see if I can replicate the failure.

wbthomason, 2020-06-08:
I've resolved this: For some reason, if `-march=native` is included in `CXXFLAGS` in the build environment, those tests run slowly.

Would you like me to open a new issue for this, or just leave it as it is?

mamoll, 2020-06-09:
Yes, please open an issue. Please include OS version, Boost version, Eigen version, and compiler version. I don't have an Arch Linux VM, but hopefully the problem is reproducible with some versions of Ubuntu.

wbthomason, 2020-06-09:
I've been trying to replicate this failure in CI and cannot. I tried to replicate the issue on another Arch machine, and also couldn't replicate the issue.

The difference seems to be CPU brand - both Arch machines are x86_64, but the one exhibiting the test failure is using an AMD CPU, and the one that does not exhibit the failure is using an Intel CPU.

I can still create a new issue once I have time to dig into why this might happen, but that behavior makes it seem likely that the bug is with the compiler, not OMPL.