ompl/ompl #731: install-ompl-ubuntu.sh can't get last release

JJRedmond, 2020-06-17:
Hi, developers!

I haven't could install OMPL by using install-ompl-ubuntu.sh and have gotten an error

2020-06-17 19:20:58--  https://github.com/ompl/ompl/releases/download/1.5.0/ompl-1.5.0.tar.gz
2020-06-17 19:20:59 ERROR 404: Not Found.

How can I solve it?

Thank you for reply

mamoll, 2020-06-18:
Thanks for reporting this issue! Please download the script again and try to run it again. With the switch from bitbucket to GitHub the URLs did not get updated correctly. Apologies for the inconvenience.

closed 2020-06-18 by mamoll

JJRedmond, 2020-06-18:
Hi, @mamoll!

Thanks for the answer.
Unfortunately script is not working yet.
The new version give this error:
./install-ompl-ubuntu.sh: line 80: syntax error near unexpected token `else'

JJRedmond, 2020-06-18:
This changes make it works

        if [ -z $APP ]; then
            wget -O - https://github.com/ompl/${OMPL}/archive/1.5.0.tar.gz | tar zxf -
            cd ${OMPL}**-1.5.0**
        else
            wget -O - https://github.com/ompl/${OMPL}/releases/download/1.5.0/${OMPL}-1.5.0-Source.tar.gz | tar zxf -
            cd $OMPL-1.5.0-Source
        **fi**

mamoll, 2020-06-18:
That's what I get for trying to fix things quickly. Thanks for the fix.

commit 5756b6a5 (2020-06-18): fix install script. closes #731.