ompl/ompl #733: RRT & RRTStar ignore goal orientation in app examples

v4hn, 2020-06-22:
I played around with the OMPL.app (release 1.5.0) a bit, after going through with a full source build of everything required to get the GUI to be available... this is really a hassle. Anyway, it's a cool GUI, though a lot of obvious improvements come to mind (solution slider, feedback on whether the planner *found* a solution, ...).

Playing with the 2d Barriers_easy example provided in the resources, I noticed that while PRM and other planners generate solutions that have a smooth connection between the last waypoint and the goal, the solutions of RRT and RRT* seem to ignore the orientation:

RRT* solution with last waypoint before goal:

[image]

PRM for comparison:
[image]

In this environment, the solution provided by RRT* seems to ignore the orientation and it looks like straight interpolation to the goal would collide with the wall.

Did I misconfigure something or is this a bug somewhere?

ChamzasKonstantinos, 2020-07-01:
Hi @v4hn, I believe this is a problem with the configuration space scaling rather than a bug of RRT/RRT*. The configuration space in this case is SE2 a.k.a (x, y ,theta) and the bounds are  0 < x < 661.02 , -440<y<0.49 (see [here](https://github.com/ompl/omplapp/blob/master/resources/2D/Barriers.cfg))  and  the biggest distance in theta is 3.14 (pi). In OMPL the collision checking resolution is defined over the statespace's maximum extend (maximum distance that exists)  see [here](http://ompl.kavrakilab.org/classompl_1_1base_1_1SpaceInformation.html).  However, because the translations maximum extends  related to the rotation's maximum extend is much bigger, the collision checking in the theta dimension will be very sparse resulting in cases were corners are cut. One solution is to increase the collision checking but the best one would be to re-scale the translation dimensions.

v4hn, 2020-07-02:
Thanks for the feedback!

I played around a bit more with the longest_valid_segment_fraction (which really should not be called "resolution" in the GUI, as a higher "resolution" actually makes checks more coarse) and RRT's range parameter.

After some more debugging it turns out that the solver indeed did not manage to solve the problem, but returned an approximate solution that got damn close. With RRT's method and without huge orientation scaling it's almost impossible to to find a path that moves into the narrow corridor with a different orientation once one such path exists already.

So let's call this issue a bug report saying that there is nothing indicating whether the shown path is an approximate or exact solution.

On top of that, to improve the chances to solve such problems, it would make sense to add an explicit SE2(/SE3) orientation/position scaling factor.

closed 2023-01-19 by zkingston