ompl/ompl #734: AssertionError running planner in blender with ompl_addon.py

RamsesReyes, 2020-06-25:
Hi, I'm trying to run the ompl add on planner in blender following the tutorial https://ompl.kavrakilab.org/morse.html. I set the environment, added the robot indicated there, defined the goal pose and the bounds, but when I run the planner, two windows appears and dissapears in a few seconds and I got the next output in the terminal after morse initialization:

**Read prefs: /home/ram/.config/blender/2.79/config/userpref.blend
OMPL: Inferred position bounds [[-55.097139,59.103779],[-57.759512,55.025254],[-5553.461473,573.838501]]
Info:    Using default state validity checker for MORSE
Info:    Registering MorseProjection as default projection evaluator for MORSE
Info:    Using the initial state of MORSE as the starting state for the planner
Writing: /tmp/blender.crash.txt
Python script error - object 'Scene_Script_Holder', controller 'Admin':
[ERROR][MORSE] Uncaught exception, quit Blender.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/morse/blender/main.py", line 734, in simulation_main
    persistantstorage.morse_services.process()
  File "/usr/lib/python3/dist-packages/morse/core/services.py", line 107, in process
    instance.process()
  File "/usr/lib/python3/dist-packages/morse/core/request_manager.py", line 346, in process
    self.main()
  File "/usr/lib/python3/dist-packages/morse/middleware/socket_request_manager.py", line 143, in main
    assert(raw[-1] == '\n')
AssertionError
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/morse/blender/main.py", line 734, in simulation_main
    persistantstorage.morse_services.process()
  File "/usr/lib/python3/dist-packages/morse/core/services.py", line 107, in process
    instance.process()
  File "/usr/lib/python3/dist-packages/morse/core/request_manager.py", line 346, in process
    self.main()
  File "/usr/lib/python3/dist-packages/morse/middleware/socket_request_manager.py", line 143, in main
    assert(raw[-1] == '\n')
AssertionError**

I'm working on Debian 10 and python 3.7. I would appreciate any help, thanks in advance. 
Ramsés Reyes

RamsesReyes, 2020-06-25:
I noticed that the socket request manager expects to receive a message ending with '\n', which apparently is not happening. So, I tried to add the '\n' at the end of the message send by the socket in line 130 of script environment.py, in such a way the message sent is 'id simulation reset_objects\n' instead 'id simulation reset_objects'. At doing this, I got the following output:

**Read prefs: /home/ram/.config/blender/2.79/config/userpref.blend
OMPL: Inferred position bounds [[-55.097139,59.103779],[-57.759512,55.025254],[-5553.461473,573.838501]]
Info:    Using default state validity checker for MORSE
Info:    Registering MorseProjection as default projection evaluator for MORSE
Info:    Using the initial state of MORSE as the starting state for the planner
Writing: /tmp/blender.crash.txt
Python script error - object 'Scene_Script_Holder', controller 'Admin':
[ERROR][MORSE] Uncaught exception, quit Blender.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/morse/blender/main.py", line 734, in simulation_main
    persistantstorage.morse_services.process()
  File "/usr/lib/python3/dist-packages/morse/core/services.py", line 107, in process
    instance.process()
  File "/usr/lib/python3/dist-packages/morse/core/request_manager.py", line 346, in process
    self.main()
  File "/usr/lib/python3/dist-packages/morse/middleware/socket_request_manager.py", line 175, in main
    is_sync, value = self.on_incoming_request(component, service, params)
  File "/usr/lib/python3/dist-packages/morse/core/request_manager.py", line 259, in on_incoming_request
    values = method(\*params) if params else method() #Invoke the method with unpacked parameters
  File "/usr/lib/python3/dist-packages/morse/services/supervision_services.py", line 70, in reset_objects
    main_reset(contr)
  File "/usr/lib/python3/dist-packages/morse/blender/main.py", line 833, in reset_objects
    b_obj.suspendDynamics()
RuntimeError: KX_GameObject.SuspendDynamics() - Blender Game Engine data has been freed, cannot use this python variable
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/morse/blender/main.py", line 734, in simulation_main
    persistantstorage.morse_services.process()
  File "/usr/lib/python3/dist-packages/morse/core/services.py", line 107, in process
    instance.process()
  File "/usr/lib/python3/dist-packages/morse/core/request_manager.py", line 346, in process
    self.main()
  File "/usr/lib/python3/dist-packages/morse/middleware/socket_request_manager.py", line 175, in main
    is_sync, value = self.on_incoming_request(component, service, params)
  File "/usr/lib/python3/dist-packages/morse/core/request_manager.py", line 259, in on_incoming_request
    values = method(*params) if params else method() #Invoke the method with unpacked parameters
  File "/usr/lib/python3/dist-packages/morse/services/supervision_services.py", line 70, in reset_objects
    main_reset(contr)
  File "/usr/lib/python3/dist-packages/morse/blender/main.py", line 833, in reset_objects
    b_obj.suspendDynamics()
RuntimeError: KX_GameObject.SuspendDynamics() - Blender Game Engine data has been freed, cannot use this python variable***
 
Thanks in advance for any help.
Ramsés Reyes

mamoll, 2020-06-26:
@calebvoss it's been a while, but do you think you could have a look at this?

zkingston, 2023-02-16:
Closing due to age. Please follow up if you are still concerned with this issue.

closed 2023-02-16 by zkingston