ompl/ompl #737: 1.5.0 version make update_bindings failed

straybird-s, 2020-07-01:
I use the same compiling environment to build 1.4.2 version that works well. I don't know why that can't work with 1.5.0.

----------------------------------------- cmake ok：-----------------------------------------
(base) ➜  Release cmake ../..
-- Build type: Release
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * Python, <https://www.python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ly/github/ompl-1.5.0/build/Release

----------------------------------------- but make update_bindings failed：-----------------------------------------
Traceback (most recent call last):
  File "/home/ly/github/ompl-1.5.0/py-bindings/generate_bindings.py", line 950, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/ly/github/ompl-1.5.0/py-bindings/generate_bindings.py", line 154, in __init__
    code_generator_t.__init__(self, 'base', ['bindings/util'], replacement)
  File "/home/ly/github/ompl-1.5.0/py-bindings/ompl/bindings_generator.py", line 146, in __init__
    self.filter_declarations()
  File "/home/ly/github/ompl-1.5.0/py-bindings/generate_bindings.py", line 297, in filter_declarations
    return_type='::ompl::base::ProblemDefinitionPtr &').exclude()
  File "/home/ly/anaconda3/lib/python3.7/site-packages/pygccxml/declarations/scopedef.py", line 834, in member_functions
    allow_empty=allow_empty)
  File "/home/ly/anaconda3/lib/python3.7/site-packages/pygccxml/declarations/scopedef.py", line 508, in _find_multiple
    "Multi declaration query returned 0 declarations.")
RuntimeError: Multi declaration query returned 0 declarations.

mamoll, 2020-07-01:
That is very odd. There are definitely declarations that should match: https://github.com/ompl/ompl/blob/1.5.0/src/ompl/base/Planner.h#L262. I suspect that you have multiple versions of OMPL installed and somehow an older version of Planner.h is found first.

closed 2020-07-06 by straybird-s

mamoll, 2020-07-07:
@straybird-s did you manage to fix the problem? If so, how? If possible, I'd like to fix the code so that someone doesn't run into the same problem in the future.

elishafer, 2020-12-30:
I'm getting the same error message when trying to create bindings for v1.5.0.
@straybird-s, have you managed to fix this?

straybird-s, 2021-01-03:
> I'm getting the same error message when trying to create bindings for ompl v1.5.1.
> @straybird-s, have you managed to fix this?
> 
> edit: I've tried deleting my build and fast-forwarding to the latest commit in master, but it doesn't help. I'm currently reverting to version 1.4.2 as this seems to fix the problem.

I tried many ways but nothing worked. Fortunately, version 1.4.2 meets my basic needs.

Jiayuan-Gu, 2021-06-22:
I have encountered the same error, but manage to solve it. I list my solution for reference.
The issue actually comes from how I install castxml. If I completely follow the official installation script, it works. It means that if you are using ubuntu <=20, you should download a specific version of castxml rather than use apt-get.

jih189, 2022-08-12:
As what Jiayuan-Gu said, you need to use the specific version of castxml rather than using apt-get, but he did not mention how to do that. Thus, I will show my solution here. Use the following link to get the binaries of Cast XML: https://data.kitware.com/#collection/57b5c9e58d777f126827f5a1/folder/57b5de948d777f10f2696370
download the one based on your ubuntu version(I am using 18.04, so the version I need is v0.3.4)
after extract it, and put it to the home directory, and add it to the path by
"export PATH=${HOME}/castxml/bin:${PATH}"
then you can compile it successfully. Remember to clean your build directory first.

You may ask how I know which version you should use. What I did is trying all of them until it works. you will find the version that works for you.

dbdxnuliba, 2023-05-11:
@jih189 , I try your method with ,but   pyplusplus_base.log shows:

```
Segmentation fault
Traceback (most recent call last):
  File "/home/dell/ref/ompl-1.6.0/py-bindings/generate_bindings.py", line 976, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/dell/ref/ompl-1.6.0/py-bindings/generate_bindings.py", line 154, in __init__
    code_generator_t.__init__(self, 'base', ['bindings/util'], replacement)
  File "/home/dell/ref/ompl-1.6.0/py-bindings/ompl/bindings_generator.py", line 132, in __init__
    self.mb = module_builder.module_builder_t(
  File "/usr/local/lib/python3.8/dist-packages/pyplusplus/module_builder/boost_python_builder.py", line 107, in __init__
    self.global_ns = self.__parse_declarations( files
  File "/usr/local/lib/python3.8/dist-packages/pyplusplus/module_builder/boost_python_builder.py", line 154, in __parse_declarations
    decls = reader.read_files( files, compilation_mode )
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/parser/project_reader.py", line 255, in read_files
    return self.__parse_all_at_once(files)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/parser/project_reader.py", line 356, in __parse_all_at_once
    return self.read_string(''.join(header_content))
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/parser/project_reader.py", line 369, in read_string
    decls = reader.read_string(content)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/parser/source_reader.py", line 377, in read_string
    decls = self.read_file(header_file)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/parser/source_reader.py", line 303, in read_file
    return self.read_cpp_source_file(source_file)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/parser/source_reader.py", line 322, in read_cpp_source_file
    xml_file = self.create_xml_file(ffname)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/parser/source_reader.py", line 268, in create_xml_file
    raise RuntimeError(
RuntimeError: Error occurred while running CASTXML:  status:139
```
and my question is /usr/local/lib/python3.8/dist-packages/pygccxml confilcit with the version 0.4.3 in ${HOME}/castxml/bin
> As what Jiayuan-Gu said, you need to use the specific version of castxml rather than using apt-get, but he did not mention how to do that. Thus, I will show my solution here. Use the following link [...]