ompl/ompl #740: In SST, how do I get cost/length of a specific node in the solution path?

elishafer, 2020-07-25:
In SST I know how to get the length of the solution path using the following Python code:
```
pdef = ss.getProblemDefinition()
pdef.getSolutionPath().length()
```
Right now I’m doing a path replanner, so I need the length/cost of the path for a specific node.
Is it possible to extract the cost/path length for a specific node in the solution path? If so, then how would one go about doing it?

Thanks

ChamzasKonstantinos, 2020-07-25:
Hi @elishafer  If I understand correctly you want the solution cost from the start node up to a specific node, since a single node has a cost of zero. You can see the functionailty that is supported for Geometric Paths [here](http://ompl.kavrakilab.org/classompl_1_1geometric_1_1PathGeometric.html) and for Control Paths [here](http://ompl.kavrakilab.org/classompl_1_1control_1_1PathControl.html): 
A simple way to do what you are asking would be this
``` path =  self.ss_.getProblemDefinition().getSolutionPath()
  path_copy = ompl.geometric.PathGeometric(ss.getSpaceInformation())
  for s in path.getStates():
      path_copy.append(s)
      print (path_copy.length())

```

elishafer, 2020-07-26:
Thanks @ChamzasKonstantinos
So the way to get the cost from start to a node x is basically to rebuild the path from start to x.
Right now I'm using the control (not geometric) for planning. That would mean that I would have to add the controls and duration to the solution like so:
```python
path = ss.getProblemDefinition().getSolutionPath()
path_copy = oc.PathControl(ss.getSpaceInformation())
c = path.getControls()
d = path.getControlDurations()
s = path.getStates()
for x in range(path.getControlCount()):
    path_copy.append(s[x], c[x], d[x])
    print (path_copy.length())
```
This works for getting path duration so this leads to another question: The the `path_copy.length()` gives the sum of the control durations of the partial path from start to node x. How can I get the path length or path cost for the partial path? Is it at all possible to get the path cost for the partial path? (I see that in the documentation it says that cost is not implemented for control paths)

ChamzasKonstantinos, 2020-07-26:
If by cost you mean the length of the equivalent geometric path I believe  you can simply  convert the control-path to a geometric path with this [function](http://ompl.kavrakilab.org/classompl_1_1control_1_1PathControl.html#aef65c5719d57101f7733a631a5cdc61a).

mamoll, 2020-08-18:
Closing this issue. Reopen if this is still an issue.

closed 2020-08-18 by mamoll