ompl/ompl #741: Compiling Error: undefined reference to ompl::base::SpaceInformation::SpaceInformation

seanroelofs, 2020-08-03:
Hello all,

I am running into trouble when using catkin to build a package that requires OMPL. Specifically, I can not instantiate a  ob::SpaceInformationPtr. I installed OMPL using the install-ompl-ubuntu.sh script. I have attached the offending code, CMakeList, and catkin output.

'
#include <ompl/config.h>
#include <ompl/base/StateSpace.h>
#include <ompl/base/SpaceInformation.h>
#include <ompl/base/spaces/SE3StateSpace.h>

namespace ob = ompl::base;

int main(int argc, char **argv) {
    ob::StateSpacePtr space(std::make_shared<ob::SE3StateSpace>());
    ob::SpaceInformationPtr si = std::make_shared<ob::SpaceInformation>(space);
    return(0);
}'

mamoll, 2020-08-03:
Do you have multiple versions of OMPL installed? OMPL 1.5.0 should be available as a ROS package for melodic, noetic, and foxy. Do you use a different version of ROS? What happens if you uninstall OMPL and add the downloaded source directory to your catkin workspace?

seanroelofs, 2020-08-04:
I uninstalled all the OMPL libraries, reinstalled using the install script, did a catkin clean, and everything worked when I rebuilt it. I'm not certain, but I think it was the catkin clean that fixed the problem because I tried building just my package after the reinstall but before the clean and it didn't work.

mamoll, 2020-08-05:
Thanks for the additional info. I will close this for now. Please reopen if it's still not fixed.

closed 2020-08-05 by mamoll