ompl/ompl #747: Constructing default projection for a compound state space of multiple SO3 state spaces

hugohadfield, 2020-08-24:
I'm currently modelling an N-link robot arm via a compound state space constructed from N SO3 state spaces. Unfortunately currently it appears I can only use an RRT planner as no default projection is registered for this compound space object. Related to this my question are:

How do I go about constructing a reasonable default projection?
If there are default projections for all spaces that make up the compound state space does it make sense to create a default projection for the compound state that is the stacking of these component default projections?

If it does make sense to do the above would it make sense to have this as a default feature in OMPL?

hugohadfield, 2020-08-24:
It looks like there is a default projection implemented for SO3: https://github.com/ompl/ompl/blob/ea77cdc5aabefd97c2d712acc3cb272df5cf6640/src/ompl/base/spaces/src/SO3StateSpace.cpp#L333

Nonetheless this code shows a 0 for `"so3 proj " << space->hasDefaultProjection() `
```
    ob::StateSpacePtr fk_space(){
        auto fk_space = std::make_shared<ob::CompoundStateSpace>();
        for (long unsigned int i=0; i < this->robot->limb_lengths.size(); i++){
            auto space(std::make_shared<ob::SO3StateSpace>()); 
            fk_space->addSubspace(space , 1.0);
            fk_space->registerProjections();
            std::cout << "so3 proj " << space->hasDefaultProjection() << std::endl;
        
        }
        fk_space->registerProjections();
        fk_space->setup();
        fk_space->sanityChecks();
        std::cout << "compound proj " << fk_space->hasDefaultProjection() << std::endl;
        return fk_space;
    }
```

hugohadfield, 2020-08-25:
I have knocked up an implementation of this automatic compound state projection with a kinda randomised dimensionality reduction here: https://gist.github.com/hugohadfield/34566a6116cc813ce79282afbad81b2a
This solves my use case as is but I'd be happy to flesh it out a bit more into a PR if there is interest from the OMPL developers

mamoll, 2020-09-25:
There's an easier way to select a projection for state spaces (compound or otherwise):

* Use the first couple values of the vector ompl::Permutation(dims) to select the projection dimension indices.
* Use [StateSpace::getValueAddressAtIndex](http://ompl.kavrakilab.org/classompl_1_1base_1_1StateSpace.html#a3b4b4123fa98346459f434e92aad3b8f) to get access to the value of the i-th double inside a state.

zkingston, 2023-01-24:
Adding on to Mark's point, take a look at [RealVectorRandomLinearProjectionEvaluator](https://ompl.kavrakilab.org/classompl_1_1base_1_1RealVectorRandomLinearProjectionEvaluator.html) for an example of a projection that works reasonably well for high-dimensional spaces. In your case, given this space corresponds to a robot arm, you should try to make a task-informed projection such as end-effector position in workspace.

closed 2023-01-24 by zkingston