ompl/ompl #751: Initialize BIT* with a known path

barbosafs, 2020-09-22:
Hello there! 
I am working on a problem that requires the planner to replan every now and then as the robot moves. So it doesn't make sense to just run 'solve' for a little longer as the path with best cost can change completely as the robot moves along the first edge.
The way I've usually seen this solved is to have the first edge fixed (commited), then make the end state of such edge the new starting state of the search algorithm, and finally start planning again.
However, in complicated environments, the planner would benefit a lot from using the previous path as an initial guess of a solution path.
Is there a way of initializing BIT* with such a path? Or perhaps feeding in a SolutionPath after clearing previous planner?
I'm using Python3, btw.

mamoll, 2020-09-25:
The trick that I have used in the path that works for any planner is the following:

* Create [a special sampler class](http://ompl.kavrakilab.org/classompl_1_1base_1_1StateSampler.html) that is initialized with the partial path you want to reuse. This sampler will return the states of the path in order. When all the path's states have been returned, it falls back on uniform random sampling. 
* Next, create [a sampler allocator function](http://ompl.kavrakilab.org/namespaceompl_1_1base.html#a729f324a0a979bd7087e3d279bbdd77e) that returns an instance of this new sampler.
* Finally, [set the planner's sampler allocator](http://ompl.kavrakilab.org/classompl_1_1base_1_1StateSpace.html#aa88e8a85586b93866ef3b2a6fb31c114) to be the function from the previous step.

Each planner is using samples in its own way, but there's a good chance that this will help preserve the previous path.

The more difficult route is to add a BIT* constructor that takes as input a PlannerData argument and reconstructs a BIT* from the data stored in this argument. This is tricky. Look at PRM for an example on how to do this.

It _may_ be possible to do some of this in Python, but I'd strongly encourage you to make these changes in C++.

barbosafs, 2020-09-28:
Thank you very much! I will give it a try.

closed 2020-09-28 by mamoll