ompl/ompl #752: MAX_QUATERNION_NORM_ERROR too low

fferri, 2020-09-25:
When using `SO3StateSpace` it happens that the initial state is invalid due to invalid bounds:

> ompl-1.5.0/src/ompl/base/src/Planner.cpp:258: PRM: Skipping invalid start state (invalid bounds)
> ompl-1.5.0/src/ompl/geometric/planners/prm/src/PRM.cpp:463: PRM: There are no valid initial states!

We work with 32-bit floats, so the precision of our internal quaternion might not be up to the tolerance required by OMPL.

We normally modify OMPL source to raise the tolerance to 1e-6:

```
--- ompl-1.5.0/src/ompl/base/spaces/src/SO3StateSpace.cpp.orig	2020-09-25 17:25:32.000000000 +0200
+++ ompl-1.5.0/src/ompl/base/spaces/src/SO3StateSpace.cpp	2020-09-25 17:21:35.000000000 +0200
@@ -44,7 +44,7 @@

 using namespace boost::math::double_constants;

-static const double MAX_QUATERNION_NORM_ERROR = 1e-9;
+static const double MAX_QUATERNION_NORM_ERROR = 1e-6;

 /// @cond IGNORE
 namespace ompl

```

Ideally, `MAX_QUATERNION_NORM_ERROR` should be increased, or there should be some API to change it without modifying the OMPL source code.

mamoll, 2020-09-25:
In your case, is the size of double 32 bits, or is the error introduced because you convert back and forth between `float` and `double`?

fferri, 2020-09-25:
Converting back and forth