ompl/ompl #753: OMPL does not register the package with the ament index for ROS2 installation

ruffsl, 2020-10-06:
From the `CMakeLists.txt` file, it seems that while the package is installing the `package.xml` in the install share folder, it still isn't registering the package with the ament index for ROS2 installations:

https://github.com/ompl/ompl/blob/253e8c0e8d59147be71d33f1a662b933fcac6774/CMakeLists.txt#L246-L247

Without registering the package with the ament index, rosdep may fail to find the package manifest:

* https://github.com/ros-infrastructure/rosdep/issues/660
* https://app.circleci.com/pipelines/github/ros-planning/navigation2/3916/workflows/cce0929c-850c-480a-8e52-0910accded65/jobs/15097/parallel-runs/0/steps/0-114

Conventionally, the `ament_package()` macro is used to register the package:

https://index.ros.org/doc/ros2/Tutorials/Ament-CMake-Documentation/#basic-project-outline

If you'd like to avoid relying on ament, you could just touch and copy the files directly, e.g:

https://github.com/ros2/demos/blob/49b12ed953d45fad67dbb27abdf76db430777a58/demo_nodes_py/setup.py#L10-L14

SteveMacenski, 2020-10-06:
Is there a way to do that in CMake and not in the python setup files?

ruffsl, 2020-10-07:
> Is there a way to do that in CMake and not in the python setup files?

Just looking via grep.app, here is a cmake example:

* https://grep.app/search?q=share/ament_index/resource_index/packages&filter[lang][0]=CMake
* https://github.com/ros-industrial/noether/blob/80d95795af4c8bfec4ac508ef602cbeb7c157528/noether_filtering/CMakeLists.txt#L185-L187

closed 2020-10-13 by mamoll