ompl/ompl #756: Compiling ompl with cmake on windows 10

ravnicas, 2020-10-27:
As the title suggests im having trouble compiling an ompl.dll on windows 10, which i would need for V-Rep as i would like to change some constant values.

I already compiled on WSL with no issues, but this compiles to *.so.

The problem arises, because the ompl cmake file doesnt find boost libraries, which are downloaded and also build for good measure.

```
PS C:\Users\username\projects\ompl> cmake . -DBOOST_ROOT=C:\Users\username\projects\boost_1_74_0\
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19042.
-- Build type: Release
CMake Error at C:/Program Files/CMake/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:218 (message):
  Could NOT find Boost (missing: serialization filesystem system
  program_options) (found suitable version "1.74.0", minimum required is
  "1.58")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:577 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.19/Modules/FindBoost.cmake:2176 (find_package_handle_standard_args)
  CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/username/projects/ompl/CMakeFiles/CMakeOutput.log".
```

Boost libraries 1.74.0 are unzipped and also build in `\username\project\boost_1_74_0`, `Eigen-3.3.8` is also in the same folder.

neemoh, 2020-12-04:
Probably too late for you now, but in case others have this issue:

As the error says, CMake cannot find the boost libs. 
I had this problem as well with boost 1.74. It was because the naming convention for the files that FindBoost.cmake was looking for was different than the compiled boost files. Namely there were system architecture (x32, x64)in the names too. What solved this issue for me was to specify the architecture in the CMakeLists by:
`set(Boost_ARCHITECTURE "-x64")`

Not sure if this is the best practice though...

neemoh, 2020-12-04:
Again, in case others have similar issues, find_package(Python) could not find Python. I have Python 3.8 installed system-wide at default locations. so this was rather strange. What worked was find_package(Python3) but then this would break the flag PYTHON_FOUND, as it was then PYTHON3_FOUND. The solution I found was to replace:
`              find_package(Python QUIET)
`
with:
`              find_package(Python COMPONENTS Interpreter QUIET)
`

Then there was another issue in the provided FindPython.cmake in CmakeModules that the find_library and find_path failed at finding either the path and libs. The solution in my case was to modify FindPython.cmake and add 
`string(REGEX REPLACE "/python.*" "" PYTHON_PREFIX2 "${PYTHON_EXEC_}")
`
and change the line:
`"${PYTHON_PREFIX}/lib"`
to
`"${PYTHON_PREFIX}/lib" "${PYTHON_PREFIX2}/libs"`
and change the line:
`"${PYTHON_PREFIX}/include"`
to
`"${PYTHON_PREFIX}/include" "${PYTHON_PREFIX2}/include"`

I am not much familiar with Windows development. So someone please correct me if I recommended bad practices.

mamoll, 2020-12-28:
Using vcpkg to install ompl is probably easiest. I have not been able to get Python binding generation to work on Windows 10, so I wouldn't worry about anything Python related.

closed 2020-12-28 by mamoll