ompl/ompl #772: Where is ompl as a python package? make update_bindings doesn't work

PeterQFR, 2021-01-12:
Hi
I'm trying to build ompl so that I can use the omplapp.py. Unfortunately I cannot get an ompl python package to be generated. Additionally, it seems installation instructions including the README.md for this repo, plus install-ompl-ubuntu.sh are out of date.

I am on ubuntu 20.04. 

after downloading the source tarball or git pulling. 
```
mkdir -p build/Release
cd build/Release
cmake ../.. -DOMPL_BUILD_PYBINDINGS=ON
make update_bindings 
```
fails with 
make: *** No rule to make target 'update_bindings'.  Stop.

changing the make target to 
```
make py_ompl
```
results in no python packages being generated but otherwise no errors. 
after build completion running in the build/Release dir:
```
find . -iname *.py
./share/ompl/demos/RigidBodyPlanningWithODESolverAndControls.py
./share/ompl/demos/KinematicCarPlanning.py
./share/ompl/demos/PlannerData.py
./share/ompl/demos/DynamicCarPlanning.py
./share/ompl/demos/RandomWalkPlanner.py
./share/ompl/demos/OptimalPlanning.py
./share/ompl/demos/KinematicChainPathPlot.py
./share/ompl/demos/RigidBodyPlanning.py
./share/ompl/demos/Point2DPlanning.py
./share/ompl/demos/SE3RigidBodyPlanning.py
./share/ompl/demos/RigidBodyPlanningWithControls.py
./share/ompl/demos/StateSampling.py
./share/ompl/demos/SE3MultiRigidBodyPlanning.py
./share/ompl/webapp/omplweb.py
./bin/ompl_benchmark_statistics.py
./bin/ompl_app.py
```
Where is ompl python library, the omplapp.py does not run without it.

mamoll, 2021-01-23:
If `make update_bindings` fails, then it means that the dependencies for generating the bindings were not detected by cmake. Install castxml, pygccxml, and pyplusplus; remove CMakeCache.txt; and run cmake again.

PeterQFR, 2021-01-24:
Thats it, all of those packages were not present on the system. Cheers

closed 2021-01-24 by PeterQFR

hjw-1014, 2021-07-27:
@PeterQFR Hi, I also meet this error: `make: *** No rule to make target 'update_util_bindings'.  Stop.`

I followed the same installation process as well, and also install castxml, pygccxml, and pyplusplus using:
```
sudo apt-get install castxml
sudo apt install python-pygccxml
sudo apt-get install python3-pip
sudo -H pip3 install -vU pygccxml pyplusplus

```
But I still got this error when I make again, I already stuck here for three days , could you give me some hints?

Thanks in adcance.