ompl/ompl #773: undefined reference to `ompl::base::SpaceInformation::SpaceInformation(std::shared_ptr<ompl::base::StateSpace> const&)

arnauddomes, 2021-01-19:
Hello

I try to compile a repo using the OMPL library:
https://github.com/ayushgaud/path_planning

When I compile ut, in get that error:

```
[ 91%] Linking CXX executable devel/lib/path_planning/path_planning_node
[100%] Linking CXX executable devel/lib/path_planning/old_path_planning
CMakeFiles/path_planning_node.dir/src/path_planning.cpp.o: In function `planner::planner()':
path_planning.cpp:(.text._ZN7plannerC2Ev[_ZN7plannerC5Ev]+0x378): undefined reference to `ompl::base::SpaceInformation::SpaceInformation(std::shared_ptr<ompl::base::StateSpace> const&)'
path_planning.cpp:(.text._ZN7plannerC2Ev[_ZN7plannerC5Ev]+0x529): undefined reference to `ompl::base::ProblemDefinition::ProblemDefinition(std::shared_ptr<ompl::base::SpaceInformation> const&)'
CMakeFiles/path_planning_node.dir/src/path_planning.cpp.o: In function `planner::plan()':
path_planning.cpp:(.text._ZN7planner4planEv[_ZN7planner4planEv]+0x533): undefined reference to `ompl::geometric::PathSimplifier::PathSimplifier(std::shared_ptr<ompl::base::SpaceInformation> const&, std::shared_ptr<ompl::base::Goal> const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/path_planning_node.dir/build.make:539: recipe for target 'devel/lib/path_planning/path_planning_node' failed
make[2]: *** [devel/lib/path_planning/path_planning_node] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/path_planning_node.dir/all' failed
make[1]: *** [CMakeFiles/path_planning_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/old_path_planning.dir/src/old_path_planning.cpp.o: In function `plan()':
old_path_planning.cpp:(.text+0x695): undefined reference to `ompl::base::SpaceInformation::SpaceInformation(std::shared_ptr<ompl::base::StateSpace> const&)'
old_path_planning.cpp:(.text+0x7e4): undefined reference to `ompl::base::ProblemDefinition::ProblemDefinition(std::shared_ptr<ompl::base::SpaceInformation> const&)'
old_path_planning.cpp:(.text+0x8a5): undefined reference to `ompl::geometric::RRTConnect::RRTConnect(std::shared_ptr<ompl::base::SpaceInformation> const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/old_path_planning.dir/build.make:539: recipe for target 'devel/lib/path_planning/old_path_planning' failed
make[2]: *** [devel/lib/path_planning/old_path_planning] Error 1
CMakeFiles/Makefile2:540: recipe for target 'CMakeFiles/old_path_planning.dir/all' failed
make[1]: *** [CMakeFiles/old_path_planning.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

I use a master source ompl to build my ompl library.

he is the code which create the error:

```
....
    // construct an instance of  space information from this state space
	ob::SpaceInformationPtr si(new ob::SpaceInformation(space));
....
    // create a problem instance
	ob::ProblemDefinitionPtr pdef(new ob::ProblemDefinition(si)); 
....

```

ChamzasKonstantinos, 2021-01-19:
This error is usually due to not explicitly inlcuding the header files that you need because many things are forward declaractions in OMPL you get linking errors and not compilation errors,  try including these headers:  

`#include "ompl/base/SpaceInformation.h"
`
`
 #include "ompl/base/ProblemDefinition.h" 
`

mamoll, 2021-01-20:
When you look at the output of `make VERBOSE=1`, do you see that the target that fails to compile is being linked against libompl? If not, what is the output of `cmake .` if you add `message("ompl llibraries: ${OMPL_LIBRARIES}")` to your CMakeLists.txt file?

Shirui-Lyu-eric, 2021-07-05:
Hi I had a similar error message(it was StateSpace which was undefined for me) working with OMPL library and this issue solved my problem "https://github.com/ompl/ompl/issues/275"
I think the problem is we have to manually instruct the linker to use the ompl library, hope this link is useful for you as well!

man-do, 2021-11-26:
@Shirui-Lyu-eric how did you solve it ???

zkingston, 2023-02-16:
Closing this due to age. It appears to be an issue with adding the OMPL libraries to the set of libraries to link against.

closed 2023-02-16 by zkingston

furkansariyildiz, 2023-11-02:
> ${OMPL_LIBRARIES}

> Closing this due to age. It appears to be an issue with adding the OMPL libraries to the set of libraries to link against.

> @Shirui-Lyu-eric how did you solve it ???

Hello @man-do,

You should add **_${OMPL_LIBRARIES}_** to your CMakeList.txt 

I am using ROS env. and I added this library to target_link_libraries part. I hope this step will solve your problem.