ompl/ompl #774: Segmentation fault (core dumped) when setup planner

xav12358, 2021-01-19:
Hello,

I try to repo for 3D path planning:
 https://github.com/abhileshborode/Motion-Planning
https://github.com/ayushgaud/path_planning

On both project, when I try to setup the planner, there is a Semgnetation fault (core dumped).

Is there a way to debug ompl to find out where this crash comes?

mamoll, 2021-01-23:
Can you create a minimal working example that triggers the seg fault? Which OS (including version)? Which version of OMPL?

caiofis, 2021-02-10:
I am having a seg fault on Ubuntu 18, with OMPL 1.2.1 when running the the first tutorial ([Geometric Planning for a Rigid Body in 3D](https://ompl.kavrakilab.org/geometricPlanningSE3.html))
Debugging the program I found that the seg fault is occur when it tries to instantiate a  ScopedState on the following line:
```C++
     ob::ScopedState<> start(space);
```

ajkxyz, 2021-11-10:
I had the same problem with `libompl-dev` installed from Ubuntu 18.04 repository.
If uninstall it with `sudo apt-get purge libompl-dev libompl12` and install latest version with the installation script https://ompl.kavrakilab.org/installation.html the problem disappears.

zkingston, 2021-11-10:
You should verify that you do not have multiple versions of OMPL installed, as there is OMPL available as `libompl<num>` / `libompl-dev` available by default and `ros-<distro>-ompl` available through ROS. Additionally, if you have OMPL installed through source either as a Catkin package or via the installation script, you should check to make sure all things that depend on OMPL are referring to the correct instance of OMPL. It is common to get conflicts if you have many versions installed, and the runtime library is not the same as the one used at compilation.

closed 2021-11-10 by zkingston

zhaianbang, 2025-01-06:
```
ompl的segment fault的问题，解决方案，已解决：
运行genesis，直接ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl，运行grasp_bottle程序出现ompl的segment fault的问题，解决步骤如下：
（重点：注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决）
用python=3.10的conda环境：
conda create genesis0106 python=3.10

conda activate enesis0106

pip install torch==2.5.1 torchvision torchaudio --index-url https://download.pytorch.org/whl/cu118

sudo apt-get -y install g++ cmake pkg-config libboost-serialization-dev libboost-filesystem-dev libboost-system-dev libboost-program-options-dev libboost-test-dev libeigen3-dev libode-dev wget libyaml-cpp-dev
export CXX=g++
export MAKEFLAGS="-j `nproc`"
pip install -vU https://github.com/CastXML/pygccxml/archive/develop.zip pyplusplus
sudo apt-get -y install castxml libboost-python-dev libboost-numpy-dev pypy3

git clone --recurse-submodules https://github.com/ompl/ompl.git -b 1.6.0
mkdir -p ompl/build
cd ompl/build

cmake .. -DPYTHON_EXEC=$(which python) -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" -DPYTHONPATH=$CONDA_PREFIX/lib/python3.10/site-packages

注意查看/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/lib中没有编译出来的4个包的报错log（_base.so _control.so _geometric.so _util.so），然后对应去解决：
/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/build/pyplusplus_util.log

如果出现cannot import name 'Protocol' from 'typing'错误：
找到对应的行，然后注释掉typing，替换加入from typing_extensions import Protocol
pip install typing_extensions==4.5.0
pip install pygccxml==2.1.0

修改注释/media/ascend/diskb1/zab2025/genesis_code/Genesis/ompl/py-bindings/generate_bindings.py中194行的：

    # self.ompl_ns.class_(f'SpecificParam< {self.string_decl} >').rename('SpecificParamString')
    try:
        self.ompl_ns.class_(f'SpecificParam< std::string >').rename('SpecificParamString')
    except:
        self.ompl_ns.class_(f'SpecificParam< std::basic_string< char > >').rename('SpecificParamString')

然后执行：
make update_bindings
（如果出现No rule for update_bindings报错，执行conda install -c conda-forge boost boost-cpp）

再执行：
make -j8
sudo make install

欢迎交流ab_zhai@foxmail.com
```