ompl/ompl #775: Incorrect package.xml file

lepalom, 2021-02-19:
ompl 1.5.2 has wrong dependencies defined in package.xml. 

* opende and libflann-dev must go to

<build_depend> section AND <exec_depend>

Because then, when the package is build they are not loaded, so are not detected, so their extensions are not built. Both dependencies must go to build_depend.

As a comment, flann is not an exec dependency, in terms of binary dependency. Checking the code, I understand that it use the flann.h file, but in a template way, so you can link against it, but there's no  binary symbol is used. In the debian package, ldd shows:

linux-vdso.so.1 (0x00007ffd293e3000)
libode.so.8 => /lib/x86_64-linux-gnu/libode.so.8
libboost_serialization.so.1.67.0 =>
libboost_filesystem.so.1.67.0 =>
libboost_system.so.1.67.0 =>
libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0
libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6
libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6
/lib64/ld-linux-x86-64.so.2 (0x00007f1354a81000)
libccd.so.2 => /lib/x86_64-linux-gnu/libccd.so.2
librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1

no /usr/lib/x86_64-linux-gnu/libflann.so.1.9.1 is showed.

However, it's necessary because there's an include file that need it.

mamoll, 2021-03-10:
What is Debian policy in cases of dependencies like libflann-dev? Make it only a build dependency? Note that OMPL's config.h will have `#define OMPL_HAVE_FLANN 1` if libflann-dev is marked as a build dependency, so it'd be nice if the header and binary release agreed on whether FLANN is present.

lepalom, 2021-03-10:
> What is Debian policy in cases of dependencies like libflann-dev?

You only need it to build applications that links against libompl-dev. But, when you have it built, there's no need to carry libflann, that was I realize checking the binary. In any case, the package.xml of the OSRF package is wrong, and it's not doing what it says that do (build with ode extension)

mamoll, 2021-03-11:
> the package.xml of the OSRF package

Which package.xml is this? is this different from the one in the OMPL repo?

zkingston, 2023-05-26:
The `package.xml` bundled with OMPL has these dependencies listed. Also, ODE has been removed as a dependency by #1050.

closed 2023-05-26 by zkingston