ompl/ompl #779: BIT* is failing when added to a benchmark

jediofgever, 2021-03-01:
Hello @mamoll ,

I am doing some study for bench-marking of optimizing planners. 
I am getting random behavior from BIT* planner. It sometimes crashes with Seg fault. 
I couldn't understand much as I don't know about implementation details.

Please refer to gdb back trace ;
[image]

I am using source build of master at  https://github.com/ompl/ompl/commit/25c76431cef25f0100ed74d09dd88944ecca5ee1. 

Please do let me know if more info is required. In order to reproduce this adding a BIT* to bench marking should suffice but as I said I get this pretty random, so I don't know if it as actual issue.

jediofgever, 2021-03-01:
not sure but seems that the seg fault is triggered when `displayProgress` is set to true in benchmark request. 

```
  ompl::tools::Benchmark::Request request(planner_timeout_, max_memory_, num_benchmark_runs_);
  request.displayProgress = true;
```
This makes sense because 

https://github.com/ompl/ompl/blob/25c76431cef25f0100ed74d09dd88944ecca5ee1/src/ompl/geometric/planners/informedtrees/src/BITstar.cpp#L121

`currentVertexProgressProperty()` is root function of this problem, and I think this function is used to track progress ? I commented out the call in that function with hope that that information is only used to track progress but I think this issue still needs deeper look

marlinstrub, 2021-03-01:
Hi @jediofgever,

Thanks for reporting this, I'll take a look. It seems that you already narrowed the problem to a specific function call, which is helpful. If you could also provide an MWE that reproduces the problem (preferably with a fixed seed for the random number generator), that would be even better.

jediofgever, 2021-03-01:
Hi @marlinstrub ,
thanks for catch up. 
At the moment unfortunately cant provide a fully compilable MWE, my code is tightly coupled with ROS and other libraries.   If you use source build of ompl(hopefully) , 
you may just add BIT* to this  bench-marking demo

https://github.com/ompl/ompl/blob/25c76431cef25f0100ed74d09dd88944ecca5ee1/demos/CForestCircleGridBenchmark.cpp#L137

 benchmark example to reproduce issue. 

The full benchmark example is here;
 https://github.com/ompl/ompl/blob/master/demos/CForestCircleGridBenchmark.cpp

marlinstrub, 2021-03-01:
Thanks @jediofgever, I'll see whether I can reproduce the issue.

marlinstrub, 2021-03-01:
@jediofgever I think the issue is likely that your application is multithreaded and accesses BIT*'s planner progress properties on one thread while BIT* is running on another (as in the benchmark you linked).

The reason this is a problem is that BIT* prunes states that are no longer useful. If it prunes a state that is being accessed through a call to the planner properties that can result in a segfault. I'll have to think about a proper fix for this but in the meantime an easy fix for you is to turn off pruning in BIT*. So instead of
```c++
b.addPlanner(std::make_shared<og::BITstar>(ss.getSpaceInformation()));
```
you could do
```c++
auto bitstar = std::make_shared<og::BITstar>(ss.getSpaceInformation());
bitstar->setPruning(false);
b.addPlanner(bitstar);
```
Let me know if that works for you.

jediofgever, 2021-03-01:
Thanks @marlinstrub , I will try your suggestion and see whether that works as an alternative solution.

jediofgever, 2021-03-02:
@marlinstrub , I applied the suggested change, after some considerable amount of runs(10 * 4 runs) I didn't see that seg fault. anymore, thanks for your responses and I hope a PR fix could be included in future. 

Best Regards

marlinstrub, 2021-03-02:
@jediofgever Glad to hear this works for you. I have a couple of options in mind for a fix, will implement one of them when I get some time.

GuancongLuo, 2021-03-02:
Hi, I have same problem for BIT* planner.

I build project by ompl1.5.1 and moveit source code. BIT* planner successfully planned in moveit.
[image]

Then, I use moveit benchmark to test BIT* and RRT* planner performance.
`roslaunch moveit_ros_benchmarks demo_panda.launch`

But the process will stop when planning by BIT*.
[image]

Here are the parameters I set.
```
 * /moveit_run_benchmark/ompl/planner_configs/BITstarkConfigDefault/drop_unconnected_samples_on_prune: 0
 * /moveit_run_benchmark/ompl/planner_configs/BITstarkConfigDefault/samples_per_batch: 100
 * /moveit_run_benchmark/ompl/planner_configs/BITstarkConfigDefault/type: geometric::BITstar
 * /moveit_run_benchmark/ompl/planner_configs/BITstarkConfigDefault/use_graph_pruning: 0
```
Some parameters not set correctly?

marlinstrub, 2021-03-02:
@GuancongLuo What do you mean by "the process will stop"? Is there an error or does BIT* not return?

GuancongLuo, 2021-03-03:
No error log. BIT* not return.

gammell, 2021-03-03:
@GuancongLuo, if you look at your output, BIT* has found a solution and had not crashed, so this is not the same problem as discussed above. Please open a new issue instead.

Thanks!
Jon

closed 2021-03-10 by mamoll

jediofgever, 2021-03-26:
Hello  @mamoll , @gammell , @marlinstrub
As I mentioned earlier we were working on evaluating optimizing planners in OMPL. A initial version of our work is now available at ;

https://arxiv.org/pdf/2103.13666.pdf

I would really appreciate that if you could have a brief look at the paper and provide some general comments or suggestions if you have any.  if something looks substaintially wrong I would have chance to revise before submitting to an actual conference. 
Best Regards

marlinstrub, 2021-03-26:
Hi @jediofgever

Unfortunately I won't have time to read your paper in full anytime soon, but skimming it quickly I noticed that some of your plots are missing labels on the y axis, and that in some of your references algorithm names are not capitalized.
I didn't see any plots that show the initial solution times and success rates of the planners, if you can find space I think these types of plots would be a great addition. Anyway, I like that you test many planners and the variety of state spaces you test the planners on. Best of luck with your submission!

jediofgever, 2021-03-26:
@marlinstrub , thank you very much for your pointers, they are very valuable!,
 I will have another look at the generated figures and try to address the naming of planners.
 There definitely could be more metrics to assess performances of planners, but unfortunately due to 6 pages limit we had squash the figures and metrics to minimum. The initial solution times are not really included in current version but the success rates could be implicitly interpreted from the figures listed as `status`. for instance the figure below shows status of acquired results after 100 runs from planners both in REEDS-SHEEP and DUBINS spaces ; 
 
 
[image]

again thank you very much for taking time to provide your thoughts. 
Best Regards

Jaeyoung-Lim, 2023-01-15:
@marlinstrub I can also confirm this issue, where BIT* occasionally segfaults when running as part of a OMPL benchmark, and it doesn't seem to be specific to the application @jediofgever is running.

marlinstrub, 2023-01-15:
@Jaeyoung-Lim Thanks for reporting. Any chance you could provide an MWE?

Jaeyoung-Lim, 2023-01-15:
@marlinstrub This is the backtrace with pruning enabled
```
#0  0x00007ffff7a6cb64 in ompl::geometric::BITstar::Vertex::isRoot() const () from /usr/local/lib/libompl.so.16
#1  0x00007ffff7a6cb8d in ompl::geometric::BITstar::Vertex::isInTree() const () from /usr/local/lib/libompl.so.16
#2  0x00007ffff7a56920 in ompl::geometric::BITstar::ImplicitGraph::numVertices() const () from /usr/local/lib/libompl.so.16
#3  0x00007ffff7a80b59 in ompl::geometric::BITstar::currentVertexProgressProperty[abi:cxx11]() const () from /usr/local/lib/libompl.so.16
#4  0x00007ffff7a80bc5 in std::_Function_handler<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (), ompl::geometric::BITstar::BITstar(std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)::{lambda()#6}>::_M_invoke(std::_Any_data const&) () from /usr/local/lib/libompl.so.16
#5  0x00007ffff7c323d7 in ompl::tools::RunPlanner::collectProgressProperties(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > ()>, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > ()> > > > const&, std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&) () from /usr/local/lib/libompl.so.16
#6  0x00007ffff74c6de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7079609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007ffff7302133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

Jaeyoung-Lim, 2023-01-15:
@marlinstrub Updating `ompl` to a more recent version (specifically `713be56e5b3ba5742d480160aee1d8d05c9551e0`) which includes https://github.com/ompl/ompl/pull/784 solved the problem.

For me it works for both turning on and off pruning 

Thanks!

marlinstrub, 2023-01-16:
@Jaeyoung-Lim Glad it's fixed, thanks for letting me know.

PR #784 (2021-03-10): Clean up BIT* planner progress properties
The main purpose of this PR is to remove the planner progress properties in BIT* that are not thread-safe (namely getting number of current vertices in graph and getting number of total samples). This should fix #779, @jediofgever can you please confirm?

I also changed the default name of BIT* from `"BITstar"` to `"kBITstar"` since by default BIT* is instantiated with a k-nearest connection model (since c08da89), which silences a warning. I also took the chance to clean up some code as it came up.