ompl/ompl #781: OMPL app visualization issue

miranisaurabh, 2021-03-02:
I was trying to visualize paths for the Home environment and noticed that the start and goal happen to be different than expected. 
Example path:
```
0.0 -145.03 70.2 0.227138573674 -0.307113663148 0.366266006751 0.848491884791
0.0 -145.03 70.2 0.227138573674 -0.307113663148 0.366266006751 0.848491884791
0.0 -145.03 70.2 0.227138573674 -0.307113663148 0.366266006751 0.848491884791
```
I expect just one single point in SE3 but observed two points (see below)
[image]

After digging into the source code and visualizing I found out that Start and Goal poses are in ZYX Euler format and the path is in XYZ Euler format.
See the difference in Start and Goal below in the Problem window: It's just ZYX->XYZ
[image]

YuhengZhi, 2021-03-03:
I can confirm this problem and can reproduce the above bug using the openPath API.

YuhengZhi, 2021-03-03:
[This (simple) pull request](https://github.com/ompl/omplapp/pull/7) should address it.

mamoll, 2021-03-11:
Fixed by merging @YuhengZhi's PR in the omplapp repo.

closed 2021-03-11 by mamoll