ompl/ompl #785: Plannerarena docker unable to change maximum upload database size

captain-yoshi, 2021-03-04:
The default max upload size is currently set to 50MB. I have tried changing the `OMPL_PLANNERARENA_MAX_DB_SIZE` [environment variable](https://github.com/ompl/ompl/blob/master/scripts/plannerarena/global.R#L41) with another value but I'm still caped to 50MB.

```sh
# Also tried with mmoll/plannerarena docker image
docker run --env OMPL_PLANNERARENA_MAX_DB_SIZE=999999999 --rm -p 8888:8888 kavrakilab/plannerarena
```
The environment value changes correctly in the docker but the local webpage halts me when uploading a db of 57MB.

[image]

captain-yoshi, 2021-03-05:
I think this environment variable must be set when building the docker. It works now.

If my assumption is right, maybe it would be good to update the docker section on this [webpage](https://ompl.kavrakilab.org/plannerarena.html) to say that you must build the docker yourself with the right environment variable to bypass the upload size limit.

mamoll, 2021-03-11:
I documented this limitation in ec3c3e3d0c24d5005ed5034e2b5db963c51f8e54. Thanks for investigating this.

closed 2021-03-11 by mamoll