ompl/ompl #786: OMPL app is not installed with ubuntu 16

kappa95, 2021-03-05:
I have installed the package as in the guide using the parameter `--app` but it returns this error: 
`Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 655, in _build_master
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 963, in require
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 854, in resolve
pkg_resources.VersionConflict: (pip 8.1.1 (/usr/lib/python3/dist-packages), Requirement.parse('pip==21.0.1'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/pip", line 6, in <module>
    from pkg_resources import load_entry_point
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 3019, in <module>
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 3003, in _call_aside
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 3032, in _initialize_master_working_set
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 657, in _build_master
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 670, in _build_from_requirements
  File "/usr/local/lib/python3.5/dist-packages/setuptools-33.1.1-py3.5.egg/pkg_resources/__init__.py", line 849, in resolve
pkg_resources.DistributionNotFound: The 'pip==21.0.1' distribution was not found and is required by the application`

mamoll, 2021-03-11:
Perhaps your PATH / PYTHONPATH settings are inconsistent with each other: `pip` is found in `/usr/lib`, but you also have it installed in `/home/kappa95/.local`. What happens if you replace the `pip${PYTHONV}` command in the script with `python${PYTHONV} -m pip`?

kappa95, 2021-03-11:
> Perhaps your PATH / PYTHONPATH settings are inconsistent with each other: `pip` is found in `/usr/lib`, but you also have it installed in `/home/kappa95/.local`. What happens if you replace the `pip${PYTHONV}` command in the script with `python${PYTHONV} -m pip`?

I tried as you adviced and it worked! The installation lasts like 6 hours but it seems works! Thank you so much!

closed 2021-03-11 by kappa95