ompl/ompl #788: Installs library with a mismatching SONAME

yurivict, 2021-03-05:
ompl installs these shared libraries:
```
lib/libompl.so
lib/libompl.so.1.5.2
lib/libompl.so.16
```

```16``` should match ```1``` in ```1.5.2```, otherwise this can confuse some toolchains.

mamoll, 2021-03-11:
Can you give an example of what breaks with the current naming scheme?

yurivict, 2021-03-11:
No, but in all examples in docs and explanations online numbers match, for example:

* https://wiki.linuxquestions.org/wiki/Library#soname
* https://stackoverflow.com/questions/12637841/what-is-the-soname-option-for-building-shared-libraries-for
* https://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html

It also appears natural that in ```libxx.so.A -> libxx.so.A.B.C``` ```.B.C``` are sort-of a middle and minor versions of ```A.B.C```.

Also in a random fragment of /usr/local/lib numbers match in SONAMEs:
```
lrwxr-xr-x    1 root  wheel        28 Feb 25 06:30 libxmlsec1-openssl.so -> libxmlsec1-openssl.so.1.2.29
lrwxr-xr-x    1 root  wheel        28 Feb 25 06:30 libxmlsec1-openssl.so.1 -> libxmlsec1-openssl.so.1.2.29
-rwxr-xr-x    1 root  wheel    247696 Feb 25 06:30 libxmlsec1-openssl.so.1.2.29
-rw-r--r--    1 root  wheel    810436 Feb 25 06:30 libxmlsec1.a
lrwxr-xr-x    1 root  wheel        20 Feb 25 06:30 libxmlsec1.so -> libxmlsec1.so.1.2.29
lrwxr-xr-x    1 root  wheel        20 Feb 25 06:30 libxmlsec1.so.1 -> libxmlsec1.so.1.2.29
-rwxr-xr-x    1 root  wheel    417376 Feb 25 06:30 libxmlsec1.so.1.2.29
-rw-r--r--    1 root  wheel      4754 Dec 11 17:16 libxshmfence.a
lrwxr-xr-x    1 root  wheel        21 Dec 11 17:16 libxshmfence.so -> libxshmfence.so.1.0.0
lrwxr-xr-x    1 root  wheel        21 Dec 11 17:16 libxshmfence.so.1 -> libxshmfence.so.1.0.0
-rwxr-xr-x    1 root  wheel     14792 Dec 11 17:16 libxshmfence.so.1.0.0
drwxr-xr-x    2 root  wheel       512 May  6  2016 libxslt-plugins
-rw-r--r--    1 root  wheel    417180 Dec 11 17:29 libxslt.a
lrwxr-xr-x    1 root  wheel        17 Dec 11 17:29 libxslt.so -> libxslt.so.1.1.34
lrwxr-xr-x    1 root  wheel        17 Dec 11 17:29 libxslt.so.1 -> libxslt.so.1.1.34
-rwxr-xr-x    1 root  wheel    257744 Dec 11 17:29 libxslt.so.1.1.34
-rw-r--r--    1 root  wheel   1116546 Dec 11 17:26 libxvidcore.a
lrwxr-xr-x    1 root  wheel        18 Dec 11 17:26 libxvidcore.so -> libxvidcore.so.4.3
lrwxr-xr-x    1 root  wheel        18 Dec 11 17:26 libxvidcore.so.4 -> libxvidcore.so.4.3
-rwxr-xr-x    1 root  wheel    800712 Dec 11 17:26 libxvidcore.so.4.3
-rw-r--r--    1 root  wheel     57616 Dec 12 00:43 libxxhash.a
lrwxr-xr-x    1 root  wheel        18 Dec 12 00:43 libxxhash.so -> libxxhash.so.0.8.0
lrwxr-xr-x    1 root  wheel        18 Dec 12 00:43 libxxhash.so.0 -> libxxhash.so.0.8.0
-rwxr-xr-x    1 root  wheel     51656 Dec 12 00:43 libxxhash.so.0.8.0
lrwxr-xr-x    1 root  wheel        12 Feb 25 00:42 libyajl.so -> libyajl.so.2
lrwxr-xr-x    1 root  wheel        16 Feb 25 00:42 libyajl.so.2 -> libyajl.so.2.1.0
-rwxr-xr-x    1 root  wheel     37040 Feb 25 00:42 libyajl.so.2.1.0
```