ompl/ompl #793: Release for ROS2 Rolling

ruffsl, 2021-03-24:
Would it be possible to release ompl for the ros2 rolling distro? Nav2 is looking into targeting rolling for our CI images, and being able to run CI jobs using a pre-built ompl package could save downstream build time.

https://discourse.ros.org/t/ros-2-rolling-distribution-plans/13227

ruffsl, 2021-05-14:
@SteveMacenski , I think this is the only package that's missing from rolling, such that we could really simplify our underlay workspace for development.

SteveMacenski, 2021-05-17:
@zkingston would this be something you'd be open to doing? It would help projects like MoveIt and Nav2 dramatically speed up our CI as well as automatically handle releases for OMPL into new ROS2 distributions without your need to release each time

mamoll, 2021-05-19:
@SteveMacenski @ruffsl I bloomed the Galactic and Rolling release of ompl a while back; see https://github.com/ros-gbp/ompl-release

closed 2025-10-01 by saahu27