ompl/ompl #799: pip._internal.exceptions.InstallationError

NayOoLwin5, 2021-04-14:
I tried installing the script with python bindings. But it threw the following errors

At first
```
Running command python setup.py egg_info
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/local/lib/python3.8/site-packages/setuptools/__init__.py", line 20, in <module>
        from setuptools.dist import Distribution, Feature
      File "/usr/local/lib/python3.8/site-packages/setuptools/dist.py", line 35, in <module>
        from setuptools import windows_support
      File "/usr/local/lib/python3.8/site-packages/setuptools/windows_support.py", line 2, in <module>
        import ctypes
      File "/usr/local/lib/python3.8/ctypes/__init__.py", line 7, in <module>
        from _ctypes import Union, Structure, Array
    ModuleNotFoundError: No module named '_ctypes'
```
It threw this error at last..
```
pip._internal.exceptions.InstallationError: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```
Is it anything to do with pip python version?
Using - 
Ubuntu 18.04
Python 2.7
pip 20.2.3 from /home/dell/.local/lib/python3.6/site-packages/pip (python 3.6)

mamoll, 2021-04-14:
You can't mix and match different python versions (like 2.7, 3.6 and 3.8).

NayOoLwin5, 2021-04-16:
Hi @mamoll, I want to install ompl for python2.7. I have set up my pip, now `pip --version`  shows
```
pip 20.2.3 from /usr/local/lib/python2.7/dist-packages/pip-20.2.3-py2.7.egg/pip (python 2.7)
```
and
```
$ python --version
Python 2.7.17
```
But the installation script is still using pip and setuptools from python3.8 as you can see the following error -
```
Running command python setup.py egg_info
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/local/lib/python3.8/site-packages/setuptools/__init__.py", line 20, in <module>
        from setuptools.dist import Distribution, Feature
      File "/usr/local/lib/python3.8/site-packages/setuptools/dist.py", line 35, in <module>
        from setuptools import windows_support
      File "/usr/local/lib/python3.8/site-packages/setuptools/windows_support.py", line 2, in <module>
        import ctypes
      File "/usr/local/lib/python3.8/ctypes/__init__.py", line 7, in <module>
        from _ctypes import Union, Structure, Array
    ModuleNotFoundError: No module named '_ctypes'
```
I have python2.7 with pip installed in /usr/local/lib/.
If python3.8 is the one causing the error in installation, how to direct the script to /usr/local/lib/python2.7?

mamoll, 2021-04-16:
1. Make sure the environment variable PYTHONPATH doesn't include /usr/local/lib/python3.8/site-packages.
2. Try removing the ompl build directory and rerun the ompl installation script

Which OS version are you using? Which OMPL version? If you have several python versions installed, it's tricky for the installation script to get everything right. You may just want to follow the ["Linux (generic)"](http://ompl.kavrakilab.org/installation.html) installation steps rather than use the script.

NayOoLwin5, 2021-05-01:
i solved it by removing python3.8 from update-alternatives and created a /usr/bin/python3 symlink for default python3.6. 
Now using the same installation script, ompl is installed for python3.6.

closed 2021-05-01 by NayOoLwin5