ompl/ompl #801: Unable to build main branch error mutlit_pass

adamkrawczyk, 2021-04-22:
Hi, 

arch: arm32
sys: ubuntu 20
boost: Version: 1.71.0.0ubuntu2

Unfortunately, I'm facing the following issue: 

```
In file included from /usr/include/boost/graph/graphviz.hpp:31,
                 from /home/user/helper_ws/ompl/src/ompl/base/src/PlannerData.cpp:44:
/usr/include/boost/spirit/include/classic_multi_pass.hpp:11:10: fatal error: boost/spirit/home/classic/iterator/multi_pass.hpp: No such file or directory
   11 | #include <boost/spirit/home/classic/iterator/multi_pass.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated
```

I've tried reinstalling boost. 

Best

mamoll, 2021-04-22:
I see this on Ubuntu 20.04, AMD64:
```
> dpkg -S /usr/include/boost/spirit/home/classic/iterator/multi_pass.hpp 
libboost1.71-dev:amd64: /usr/include/boost/spirit/home/classic/iterator/multi_pass.hpp
```
That file doesn't exist on your machine? Can you uninstall and reinstall libboost1.71-dev to see if that fixes things?

adamkrawczyk, 2021-04-23:
$ `dpkg -S /usr/include/boost/spirit/home/classic/iterator/multi_pass.hpp`
Output: libboost1.71-dev:armhf: /usr/include/boost/spirit/home/classic/iterator/multi_pass.hpp

adamkrawczyk, 2021-04-23:
After reinstalling it works, even though as you can see I got this installed. Thanks.

closed 2021-04-23 by adamkrawczyk