ompl/ompl #802: BFMT possibly inverting path segments in MotionValidator while expanding from goal.

werner291, 2021-05-01:
Hello,

I believe I have found a possible bug in the implementation of BFMT.

When using a MotionValidator with a non-symmetric predicate, the trajectory produced by BFMT often start with valid motions, then end in a sequence of invalid motions (according to the predicate) near the end. Most notably, these motions become valid once reversed.

Is it possible that the code that expands the tree originating in the goal position is not accounting for the direction of movement when calling `MotionValidator::checkMotion`?

werner291, 2021-05-01:
I checked in https://github.com/ompl/ompl/blob/main/src/ompl/geometric/planners/fmt/src/BFMT.cpp, and there do not appear to be any checks when calling `checkMotion` for whether the motion is reversed or not.

mamoll, 2021-05-01:
Are you using a custom MotionValidator where the direction of the motion matters for its validity? Usually, motion validators just check whether a path segment is collision free and direction doesn't matter. If you have a minimal example to demonstrate the issue, that would help a lot in fixing the underlying problem.

werner291, 2021-05-02:
Indeed, the direction of motion matters in my case.

Basically, the purpose of the motion validator in my case is to ensure that the robot I have only moves in a direction that is covered by the robot's sensors, such that it can see unexpected obstacles that show up in its' path. There is therefore a significant difference in moving forward or backward.

I was just trying out several algorithms, and noticed that BFMT would produce paths that would cause the robot to stop, then reverse towards the goal, exactly in the opposite direction from which the sensors are facing.

I'll post a minimal example when I get to work sometime tomorrow, but the fix would probably be to check whether you're on the tree rooted at the start or at the goal when running the validator.

For now, here's the predicate I'm using right now:

```
bool LookForwardValidator::checkMotion(const` ompl::base::State *s1, const ompl::base::State *s2) const {

    // Custom state space, but it's basically just SE3 with restricted orientation and continuously memory
    // All that matters is that you can get position in space and a forward vector.
    auto ss1 = s1->as<PositionAndHeadingSpace::StateType>(); 
    auto ss2 = s2->as<PositionAndHeadingSpace::StateType>();

    Eigen::Vector3d forward_local(0.0, 1.0, 0.0);

    Eigen::Vector3d delta_linear(ss2->getX() - ss1->getX(), ss2->getY() - ss1->getY(), ss2->getZ() - ss1->getZ());
    double distance = delta_linear.norm();

    if (distance > std::numeric_limits<double>::epsilon()) {
        auto fwd_1 = ss1->rotation() * forward_local;
        auto fwd_2 = ss2->rotation() * forward_local;
        return super_->checkMotion(s1, s2) &&
                (fwd_1.dot(delta_linear) / distance > 0.8) && // Just check if the cosine is sufficiently close to 1
                (fwd_2.dot(delta_linear) / distance > 0.5);
    } else {
        return super_->checkMotion(s1, s2);
    }
}
```

werner291, 2021-05-02:
However, I do concede that the doc comment on the MotionValidator (https://ompl.kavrakilab.org/classompl_1_1base_1_1MotionValidator.html#a9b6d254308ea12799a72049a8dff3fc6) is not entirely clear on how to deal with forwards or backwards motion. Actually, it's a bit contradictory, since it both states "from s1 to s2", implying directionality, and that "This function assumes s1 is valid.", assuming tree expansion from `s1`.

The current implementation may be completely valid, depending on how you interpret that.

werner291, 2021-05-02:
Notice how RRTConnect does it: https://github.com/ompl/ompl/blob/main/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp#L142

```
    bool validMotion = tgi.start ? si_->checkMotion(nmotion->state, dstate) :
                                   si_->isValid(dstate) && si_->checkMotion(dstate, nmotion->state);
```

It specifically reverses the motion validator's arguments if not expanding from the start, and specifically adds an extra check for one of the states (the one assumed to be valid by the validator) when doing so.

werner291, 2021-05-03:
Here's something to test with:
```
#include <ompl/base/spaces/SE3StateSpace.h>
#include <ompl/geometric/planners/rrt/RRTConnect.h>
#include <ompl/geometric/planners/fmt/FMT.h>
#include <ompl/geometric/planners/fmt/BFMT.h>
#include <ompl/geometric/planners/bitstar/BITstar.h>
#include <ompl/base/DiscreteMotionValidator.h>
#include <ompl/base/spaces/constraint/ProjectedStateSpace.h>
#include <ompl/base/Constraint.h>
#include <algorithm>

namespace ob = ompl::base;
namespace og = ompl::geometric;

class LookForwardValidator : public ompl::base::MotionValidator {

public:
    explicit LookForwardValidator(const ompl::base::SpaceInformationPtr &si) : MotionValidator(si) {}

    bool checkMotion(const ompl::base::State *s1, const ompl::base::State *s2) const override {

        auto ss1 = s1->as<ob::SE3StateSpace::StateType>();
        auto ss2 = s2->as<ob::SE3StateSpace::StateType>();

        Eigen::Vector3d forward_local(0.0, 1.0, 0.0);

        Eigen::Vector3d delta_linear(ss2->getX() - ss1->getX(), ss2->getY() - ss1->getY(), ss2->getZ() - ss1->getZ());
        double distance = delta_linear.norm();

        if (distance > std::numeric_limits<double>::epsilon()) {

            auto fwd_1 = Eigen::Quaterniond(ss1->rotation().w, ss1->rotation().x, ss1->rotation().y, ss1->rotation().z) * forward_local;
            auto fwd_2 = Eigen::Quaterniond(ss2->rotation().w, ss2->rotation().x, ss2->rotation().y, ss2->rotation().z) * forward_local;

            return (fwd_1.dot(delta_linear) / distance > 0.5) && (fwd_2.dot(delta_linear) / distance > 0.5);
        } else {
            return true;
        }
    };

    bool checkMotion(const ompl::base::State *s1, const ompl::base::State *s2,
                     std::pair<ompl::base::State *, double> &lastValid) const override {
        abort(); // Not implemented.
    }
};

int main(int argc, char **argv) {

    ob::RealVectorBounds bounds(3);
    bounds.setLow(-10.0);
    bounds.setHigh(10.0);
    auto space = std::make_shared<ob::SE3StateSpace>();
    space->setBounds(bounds);

    auto si(std::make_shared<ob::SpaceInformation>(space));

    auto mv(std::make_shared<LookForwardValidator>(si));
    si->setMotionValidator(std::static_pointer_cast<ob::MotionValidator>(mv));

    si->setStateValidityChecker([](const ob::State *st) {
        return true;
    });
    si->setup();

    ompl::base::ScopedState<ob::SE3StateSpace> start(space);
    start->setXYZ(0.0, -5.0, 0.0);
    start->rotation().setAxisAngle(0.0,0.0,1.0,0.0);

    ob::ScopedState<ob::SE3StateSpace> goal(space);
    goal->setXYZ(0.0, 5.0, 0.0);
    goal->rotation().setAxisAngle(0.0,0.0,1.0,M_PI);

    auto pdef(std::make_shared<ob::ProblemDefinition>(si));
    pdef->setStartAndGoalStates(start, goal, 0.5);

    auto planner(std::make_shared<og::BFMT>(si));
    planner->setProblemDefinition(pdef);

    ob::PlannerStatus solved = planner->ob::Planner::solve(10.0);

    ompl::base::PlannerData pd(si);
    planner->og::BFMT::getPlannerData(pd);

    if (solved) {
        // print the path to screen
        const ob::PathPtr path = pdef->getSolutionPath();

        std::cout << "Found solution:" << std::endl;
        path->print(std::cout);

        auto states = path->as<ompl::geometric::PathGeometric>()->getStates();

        for (int i = 0; i < states.size()-1; i++) {
            printf("Motion valid? %s reverse: %s\n", mv->checkMotion(states[i],states[i+1]) ? "Yes" : "No", mv->checkMotion(states[i+1],states[i]) ? "Yes" : "No");
        }
    }

    return 0;
}
```

It gave me this output:

```
Motion valid? Yes reverse: No
Motion valid? Yes reverse: No
Motion valid? No reverse: Yes
Motion valid? No reverse: Yes
```

This means: the motion would have been valid if the last two segments had been reversed.