ompl/ompl #806: ImportError: cannot import name 'util' from 'ompl' (unknown location)

ElegantLin, 2021-05-14:
I download the omplapp-1.5.2-Source from the official website on my Mac whose version is Big Sur. I installed MacPort and followed the instructions in https://ompl.kavrakilab.org/installation.html. It is OK to `import ompl` but I failed to

```
from ompl import util
from ompl import base
```

The error is 

`cannot import name 'base' from 'ompl' (unknown location)`

I did not change anything. Could someone help me with this?

Thanks

mamoll, 2021-05-20:
It's quite likely something went wrong in the python binding generation process. It appears to be broken at the moment on macOS. There's a bug in pygccxml, a Python package used for binding generation. I create a PR with a fix [here](https://github.com/CastXML/pygccxml/pull/142).

zkingston, 2023-02-15:
Closing due to age and #807

closed 2023-02-15 by zkingston