ompl/ompl #807: Ubuntu import error

ElegantLin, 2021-05-14:
Hi, I changed to Ubuntu platform. I followed the shell instruction as stated [here](https://ompl.kavrakilab.org/installation.html). I built and installed it successfully. I met an error when I am trying to import the package

```
>>> from ompl import base
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/ompl/base/__init__.py", line 2, in <module>
    from ompl.base._base import *
ImportError: /usr/lib/python3/dist-packages/ompl/base/_base.so: undefined symbol: _ZNK5boost6python5numpy7ndarray5shapeEi
>>>
```

The is nothing error when I am trying to import util. I changed nothing in the whole process.

Could you please help me?

Thanks!

mamoll, 2021-05-14:
If you still have your build directory, can you attach the file `CMakeCache.txt` from that directory? Also, what is the output of `ldd /usr/lib/python3/dist-packages/ompl/base/_base.so`?

ElegantLin, 2021-05-15:
Yes. I have. Let me put them seperately.

The output of `ldd` 

```
mk@mk-SYS-4029GP:/data/model$ ldd /usr/lib/python3/dist-packages/ompl/base/_base.so
        linux-vdso.so.1 (0x00007ffcca530000)
        libompl.so.16 => /usr/local/lib/libompl.so.16 (0x00007f9c186e1000)
        libboost_python3-py36.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_python3-py36.so.1.65.1 (0x00007f9c184a2000)
        libpython3.6m.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0 (0x00007f9c17df7000)
        libboost_numpy3-py36.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_numpy3-py36.so.1.65.1 (0x00007f9c17bec000)
        libboost_serialization.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1 (0x00007f9c179ae000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9c17625000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9c17287000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9c1706f000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9c16c7e000)
        libode.so.6 => /usr/lib/x86_64-linux-gnu/libode.so.6 (0x00007f9c16960000)
        libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f9c16746000)
        libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f9c16541000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9c16322000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f9c1a224000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f9c160f0000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f9c15ed3000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9c15ccf000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f9c15acc000)
```

ElegantLin, 2021-05-15:
My CMakeCache.txt is

```
# This is the CMakeCache file.
# For build in directory: /data/model/ompl-1.5.2/build/Release
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/local/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy36 library (debug)
Boost_NUMPY36_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY36_LIBRARY_DEBUG-NOTFOUND

//Boost numpy36 library (release)
Boost_NUMPY36_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY36_LIBRARY_RELEASE-NOTFOUND

//Boost numpy3 library (debug)
Boost_NUMPY3_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy3.so

//Boost numpy3 library (release)
Boost_NUMPY3_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy3.so

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py36 library (debug)
Boost_PYTHON-PY36_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python-py36.so

//Boost python-py36 library (release)
Boost_PYTHON-PY36_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python-py36.so

//Boost serialization library (debug)
Boost_SERIALIZATION_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost serialization library (release)
Boost_SERIALIZATION_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost system library (debug)
Boost_SYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost system library (release)
Boost_SYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

[... 648 lines cut ...]
YAMLCPP_yaml-cpp_PREFIX:INTERNAL=
YAMLCPP_yaml-cpp_VERSION:INTERNAL=
//Components requested for this build tree.
_Boost_COMPONENTS_SEARCHED:INTERNAL=filesystem;numpy3;numpy36;program_options;python-py36;serialization;system;unit_test_framework
//Last used Boost_INCLUDE_DIR value.
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/local/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
prefix_result:INTERNAL=

```

ElegantLin, 2021-05-15:
Thanks for your help :)

zkingston, 2023-02-15:
Closing this given age - open another issue if you still cannot build bindings.

closed 2023-02-15 by zkingston