ompl/ompl #810: A question regaring to ValidStateSampler

jediofgever, 2021-05-26:
Dear @mamoll ,

I have a custom `ValidStateSampler` which samples states based on topology of map that I used to represent environment.
I found out that some of planners do not use this this custom sampler although I set it  with;

```
  simple_setup_->getSpaceInformation()->setValidStateSamplerAllocator(
    std::bind(&CompositePlanner::allocValidStateSampler, this, std::placeholders::_1));
```
For instance, RRT variants ignores custom  `ValidStateSampler` but PRMs uses the custom `ValidStateSampler`. 
Is it possible to replace default statesampler(unifrom sampler)  with my own for any Planner ? 

Thank you for your time.

mamoll, 2021-05-26:
This is actually expected behavior. Planning algorithms are not required to use a `ValidStateSampler`. RRT and its variants have a different sampling strategy that will, of course, still check the validity of states and motions it adds to the tree, but they don't need to generate random valid states.

ChamzasKonstantinos, 2021-05-26:
Depending on your use-case you might find want to directly override the [StateSampler](http://ompl.kavrakilab.org/classompl_1_1base_1_1StateSpace.html#aa88e8a85586b93866ef3b2a6fb31c114)  which I think is used by all planners, but does not necessarily produce collision free (valid) samples