ompl/ompl #811: Longest Valid Segment Length and State validity checking resolution

ksatyaki, 2021-05-28:
The description on the OMPL website seems a bit vague for me. Sorry, but as I understand, this is a percentage of the maximumExtent of the state space right? 

Therefore, if I want my path segments to be utmost 0.1 m apart (and given that I can guarantee collission-free segments of this length), `StateSpace::validSegmentCount()` should return the number 10 for two states a metre apart, correct?

Therefore, when the distances between every two path points for a particular solution is computed, these should all be the same number: the number returned by `getLongestValidSegmentLength()` correct? 

Am I missing something here? 
I can also see that `StateSpace::validSegmentCount()` returns a seemingly very random number. When I printout `space.distance(state1, state2) / space.getLongestValidSegmentLength()`, this doesn't seem to be anywhere close to what `validSegmentCount()` returns.

I am using a Dubins State Space.

ChamzasKonstantinos, 2021-05-28:
Hu @ksatyaki . Let me see if I can help.

- _this is a percentage of the maximumExtent of the state space right?_
Yes, the `longestValidSegment` is calculated as (fraction*maximum extend). this fraction is the  [longestValidSegmentFraction ](https://ompl.kavrakilab.org/classompl_1_1base_1_1StateSpace.html#aeaba86765fbfc6686312e50934edfd70)

- _Therefore, if I want my path segments to be utmost 0.1 m apart (and given that I can guarantee collission-free segments of this length), StateSpace::validSegmentCount() should return the number 10 for two states a metre apart, correct?_
This will be true if your StateSpace is R2 or R3, which you can calculate in meters.  if you have a `CompoundStateSpace` like `DubinsStateSpace` you have to take into account the other dimensions as well. 

- _Therefore, when the distances between every two path points for a particular solution is computed, these should all be the same number: the number returned by getLongestValidSegmentLength()_ correct?
 This will be true if a) the solution is fully interpolated in the end and b) the Path is a Geometric path, If you are using a Control path interpolation will not have the same effect see [here](http://ompl.kavrakilab.org/classompl_1_1control_1_1PathControl.html#aff4f8651ac48daa0d8f58fb378767305).
 
 - _I can also see that StateSpace::validSegmentCount() returns a seemingly very random number. When I printout space.distance(state1, state2) / space.getLongestValidSegmentLength(), this doesn't seem to be anywhere close to what validSegmentCount() returns._ correct?
 This would be true only for simple `StateSpaces`. For Compound state spaces (like `DubinsStateSpace`) [https://ompl.kavrakilab.org/classompl_1_1base_1_1DubinsStateSpace.html]) the getLongestValidSegmentLength() returns the max of the subspaces as documented [here](http://ompl.kavrakilab.org/classompl_1_1base_1_1CompoundStateSpace.html#a133df8e19915dd8f805a054509e38256)
 
 Let me know if this does not answer your question

ksatyaki, 2021-05-29:
Thank you so much for the clarifications. 
Looks like the only point I missed was the Orientation subspace. I thought that since the max distance between any two poses can be utmost 3.14, the number of subdivisions inside that can't be more than the subdivisions possible within a line segment. Apparently, I was wrong. 

For a dubins state space, it seems a bit wrong to compare these two and take the max, don't you think? What would be the reasoning there?

ChamzasKonstantinos, 2021-05-31:
I think this is due to the more "conservative" collision checking approach taken in the `CompoundStateSpace` implementation. I believe the main idea is that when you have  `CompoundStateSpace` you want the collision checking (thus the longestValidSegment) to be at least as detailed (small)  as the most detailed subspace because you want to avoid collisions with the resolution of each subspace. 

I think from an interoperability point of view its a bit confusing but it is more "safe" .

ksatyaki, 2021-06-03:
I understand the logic behind a CompoundStateSpace returning the values as you mentioned. However, the DubinsStateSpace can overload this function and return the slightly more practical values. :-)

zkingston, 2023-02-16:
Closing due to age. We would welcome a pull request to specify more precise values for the `DubinsStateSpace`.

closed 2023-02-16 by zkingston

rconnorlawson, 2023-03-09:
I just uncovered this bug myself while working with `DubinsStateSpace`. At least since v.1.5.2, `DubinsStateSpace` does not override `validSegmentCount`, which means the default implementation from `CompoundStateSpace` is used. 

`DubinsStateSpace` is a subclass of `SE2StateSpace`, which is itself a subclass of `CompoundStateSpace` with a two-dimensional `RealVectorStateSpace` (weight 1) and an `SO2StateSpace` (weight 0.5) as its components.

`validSegmentCount` is used for determining the number of points to use for interpolation during collision detection. In Dubins space, interpolation yields points along the Dubins curve. Therefore the relevant distance is the Dubins path length.

However, as discussed above, the default implementation of `validSegmentCount` from `CompoundStateSpace` simply delegates to each of its component spaces, return the maximum value (i.e. the value corresponding to the most detailed subspace).

A better solution for `DubinsStateSpace` is to override `validSegmentCount` with an implementation that uses the Dubins path length.

@zkingston Do you agree? I can put together a pull request.

zkingston, 2023-03-09:
Yes, I'd welcome a pull request to address the above. Overriding `validSegmentCount` with a more appropriate value makes sense.

PR #1001 (2023-03-09): Fix incorrect validSegmentCount in DubinsStateSpace
`validSegmentCount` is used to determine the number of interpolation points to use for collision checking a segment.

`DubinsStateSpace` incorrectly delegated to the default `CompoundStateSpace::validSegmentCount` implementation, which evaluates each subspace independently and returns the highest value.

This patch returns to the basic `StateSpace` behavior, computing a count based on edge length -- in this case, the length of the Dubins path.

Fixes #811.