ompl/ompl #822: How can I get the whole tree or all the information generated by the planners

hjw-1014, 2021-07-26:
Hi, I have a question but I'm so sure where can I ask for help, so I open an issue here, if it's not appropriate I will close then.

For now, I use` RRTstar ` in **MoveIt** for my robot (TIAGo++, which is a mobile manipulator) for base navigation. I wanna get the whole tree or all the information generated by the planners (In my case it's RRTstar), and then I will use this information for other random start states and generate a vector field based on the RRTstar tree. 
 
I do notice there is a function `Show Exploration` in **OMPL gui**  to show the positions of all the states generated by the planner, but I don't find any solutions that I can get the whole information if I just use MoveIt with OMPL plugin, cos I created a moveit_config for my robot, and modify the [move_froup_python_interface_tutorial.py](https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py) to plan.  

**So in my python file how can I get these paths or nodes in a matrix or .json, .csv, or in excel ?**

My Env:
**Ubuntu 18.04
ROS melodic**

Thanks in advance

zkingston, 2021-07-26:
To extract the planner graph (for any motion planner), use [Planner::getPlannerData()](https://ompl.kavrakilab.org/classompl_1_1base_1_1Planner.html#ae8af389dbcdf416e142dc2f1cf5a8829) to fill out an instance of
[PlannerData](https://ompl.kavrakilab.org/classompl_1_1base_1_1PlannerData.html). There is a demo of how to do this [in Python](https://ompl.kavrakilab.org/PlannerData_8py_source.html).

closed 2021-07-26 by zkingston

hjw-1014, 2021-07-27:
> To extract the planner graph (for any motion planner), use [Planner::getPlannerData()](https://ompl.kavrakilab.org/classompl_1_1base_1_1Planner.html#ae8af389dbcdf416e142dc2f1cf5a8829) to fill out an [...]

Thank you for your quick answer.