ompl/ompl #823: No such file or directory: 'bindings/geometric'

hjw-1014, 2021-07-28:
Hi, I wanna use OMPL in ROS and also combine it with MoveIt. I already tried installed `ompl` in melodic` sudo apt-get install ros-melodic-ompl`, and I can use OMPL as a MoveIt plugin.

But for now my problem is I wanna use my moveit python script, and get the whole planner data. I already know in OMPL I can use `getPlannerData` to do it, like[ this example ](https://ompl.kavrakilab.org/PlannerData_8py_source.html). 

So what I have done is:
1. I git clone OMPL github repository and use `catkin build `, then I found when I tried to run an example like PlannerData.py, I got the error:
```
 huangjiawei@huangjiawei-Ubuntu:~/tiago_dual_public_ws/src/ompl/demos$ rosrun ompl PlannerData.py 
[rosrun] You have chosen a non-unique executable, please pick one of the following:
1) /home/huangjiawei/tiago_dual_public_ws/devel/share/ompl/demos/PlannerData.py
2) /home/huangjiawei/tiago_dual_public_ws/src/ompl/demos/PlannerData.py
3) /home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/demos/PlannerData.py
4) /home/huangjiawei/tiago_dual_public_ws/src/ompl/demos/PlannerData.py
5) /home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/demos/PlannerData.py
#? 3
Failed to import graph-tool.  PlannerData will not be analyzed or plotted
Traceback (most recent call last):
  File "/home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/demos/PlannerData.py", line 59, in <module>
    from ompl import base as ob
  File "/home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/py-bindings/ompl/base/__init__.py", line 1, in <module>
    from ompl import util
  File "/home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/py-bindings/ompl/util/__init__.py", line 4, in <module>
    from ompl.util._util import *
ImportError: dynamic module does not define init function (init_util)
```
I might ran cmake twice stupidly, so that's why there are multiple python files.

2. And then inside ompl dictionary which is in my ros worspace, I followed here to install OMPL:
```
mkdir -p build/Release
cd build/Release
cmake ../..
# next step is optional
make -j 4 update_bindings # if you want Python bindings
```
It reported error:
```
huangjiawei@huangjiawei-Ubuntu:~/tiago_dual_public_ws/src/ompl/build/Release$ cmake ../..
-- Build type: Release
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Python, <https://www.python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following features have been disabled:

 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.

-- The following OPTIONAL packages have not been found:

 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/huangjiawei/tiago_dual_public_ws/src/ompl
huangjiawei@huangjiawei-Ubuntu:~/tiago_dual_public_ws/src/ompl/build/Release$ make -j 4 update_bindings 
make: *** No rule to make target 'update_bindings'.  Stop.
```

I already checked the issues in the last three days but I cannot figure it out, but I need to get the planner information now. If you can give me some help I would be very appreciate it. Thanks in advance.

hjw-1014, 2021-07-28:
The error occurs during I ran `./install-ompl-ubuntu.sh --python` just now. 

```
Replacing member function ::ompl::tools::Benchmark::setPostRunEvent
still alive after 60.0632 seconds!
still alive after 120.076 seconds!
still alive after 180.087 seconds!
still alive after 240.142 seconds!
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "/home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/py-bindings/generate_bindings.py", line 950, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/py-bindings/generate_bindings.py", line 845, in __init__
    replacement, 1)
  File "/home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/py-bindings/ompl/bindings_generator.py", line 149, in __init__
    self.mb.register_module_dependency(dep)
  File "/usr/local/lib/python3.6/dist-packages/pyplusplus/module_builder/boost_python_builder.py", line 142, in register_module_dependency
    db.load( other_module_generated_code_dir )
  File "/usr/local/lib/python3.6/dist-packages/pyplusplus/utils/__init__.py", line 129, in load
    f = open( fpath, 'r+b' )
FileNotFoundError: [Errno 2] No such file or directory: 'bindings/geometric'
-- Build type: Release
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Python, <https://www.python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/huangjiawei/tiago_dual_public_ws/src/ompl/ompl-1.5.2/build/Release
[100%] Built target update_tools_bindings
Scanning dependencies of target update_bindings
[100%] Built target update_bindings
Scanning dependencies of target ompl
[  0%] Building CXX object src/ompl/CMakeFiles/ompl.dir/datastructures/src/AdjacencyList
```

zkingston, 2023-01-24:
This means there was another failure of generating one of the Python modules, seems similar to #977, closing this in favor of that issue.

closed 2023-01-24 by zkingston